thonny/thonny #1054: Update German translation

aivarannamaa, 2020-01-10:
Transferred from #1051

aivarannamaa, 2020-01-10:
@Hermann12

> Some translations are not taken? e.g. "Run current script" isn't even into the .mo file why?

Strange, I found it in Poeditor and I also see it in Thonny when I update German ("Aktuelles Skript ausführen").

> Also other menue text e.g. close all -> Alle schließen? Is this because of the ä,ü,ß, ... ?

This is also nicely presented in my (Windows 10) computer.

> How can I send you the .mo and .po files?

You don't need to. Before each release I'll update the languages in Thonny, which are updated in Poeditor.

Hermann12, 2020-01-10:
Thanks, I have translated again. Now it looks better. Test is o.k. so far. Maybe read my comments on PoEdit if possible. Thanks!

PS: I have also realized, that the item under Tools | "Manage packages ..." and "Manage Plug-ins" are very similar. maybe this can be handeled by a filer or tab in the same Pop Up Window? Only a idea.

aivarannamaa, 2020-01-13:
> the item under Tools | "Manage packages ..." and "Manage Plug-ins" are very similar. Maybe this can be handeled by a filer or tab in the same Pop Up Window?

They are used on different occasions, so I think they are better as separate dialogs. BTW, they contain same packages only with "same as front-end" interpreter settings.

> Maybe read my comments on PoEdit if possible.

Done. I didn't realize it is occasionally so hard to infer the meaning of a string out of context. Thank you for asking for clarifications!

> Also the function under file shows always the same behaviour. I assumed to see [...]
You meant there should be different initial contents in the file-name box in file dialog?

Hermann12, 2020-01-13:
> You meant there should be different initial contents in the file-name box in file dialog?<
Yes, otherwise I don't know, why the menu should have 3 options.

closed 2021-01-20 by aivarannamaa