thonny/thonny #1057: Thonny opens files read-only

aivarannamaa, 2020-01-13:
https://thonny.org/blog/2019/02/09/version_3_1_1.html#comment-4752834609

Barry Fagin:
> I had IDLE previously, installed Thonny for a class. My files open in Thonny, but I can't edit them! I have to "save as" or "save a copy", those work fine. I checked, the file is not read-only, looks normal in Windows. Any ideas?

Aivar Annamaa:
> Does it mean Thonny's editor is read only or you can't save your edits? Can you edit same files with IDLE? Do you open the files with Ctrl+O / File=>Open or from Windows Explorer? Do you get any error messages?

Barry Fagin:
> Fascinating. The problem only occurs when I double-click on the file. Windows opens it with Thonny, but read-only. If I open files from within Thonny, they are editable. A minor annoyance, but I can live with it. Just an interesting quirk I guess.

WilsonRL, 2021-09-28:
I'm having the save issue and I feel like the issue is somehow related to the pop-up windows. 

If I double click to open the file or just open the file from outside of Thonny, it won't let me edit the program or enter input when I run the program. If I open the file from within Thonny, it works just fine. 

The interesting thing I noticed is that opening any pop-up window in Thonny fixes the problem. 

For example, if Thonny isn't already running and I double click to open a file, it won't let me edit or enter input when I run the program. However, if I open preferences and click okay without making any changes or if I decide to open a file from within Thonny, but then change my mind and choose cancel so that I don't actually open a file, that same program that is opened in Thonny and wasn't editable now lets me edit it and runs with no issues. On the other hand, opening the help function which appears in the Thonny editor and doesn't make a pop-up window appear doesn't fix the problem (you still can't edit the program). 

I'm using a MacBook Pro with macOS Big Sur version 11.6 in case that matters.

aivarannamaa, 2021-11-04:
@WilsonRL, many thanks for the input! And sorry for delayed reply!

This reminds me that I saw something like this on Linux while I was developing another part and tried to automatically set focus to the editor -- with certain configuration the focus became locked to some unknown place.

DavidAyliffe, 2022-08-24:
> For example, if Thonny isn't already running and I double click to open a file, it won't let me edit or enter input when I run the program. However, if I open preferences and click okay without [...]

SAME