thonny/thonny #1059: Separated shell

isanzyy, 2020-01-14:
When i start up thonny the shell opens in a new window, all black like comand prompt. If i close that window i cant run commands anymore or save them, and then it says "Backend terminated or disconnected. Use 'Stop/Restart' to restart ...".
Also when i try to use files.txt i cant write on them.
I tried other versions of python and even other programs for programming, and the same thing happens on all of them.

aivarannamaa, 2020-01-14:
* Can you use Thonny when the black window remains open?
* Which Thonny version are you using?
* Which Windows version?
* Is your Windows display language English or something else?

isanzyy, 2020-01-14:
i can but when i run any program it pops 3 of those same windows. Also when i use files and .txt i cant write on them.
i tried on the 3.2.1 version on 3.2.4 and 3.2.6 and all of them gave me the same window and same problems.
I even tried on diferent programs itself like Visual Studio and the same thing happens.
My windows is on the latest version. This started happening out of nowhere.
And my windows is in my language which is portuguese.

aivarannamaa, 2020-01-16:
I've heard of similar issues before and always these have on with non-english Windows.

So far I haven't been able to reproduce this in my machine, but I'll try some more.