thonny/thonny #1060: thonny-xxl-3.2.6 artifact contains `__pycache__` folder

stdedos, 2020-01-14:
I am not sure if it's a bug, or a feature:

I was trying to determine the differences of the released artifacts. I opened up the artifacts, and started comparing names and sizes.

I noticed that in `thonny\lib\python3.7`, there is an extra `__pycache__` folder

aivarannamaa, 2020-01-14:
Thank you for noticing this! I'll add required cleaning commands to the build script

thijstriemstra, 2020-01-15:
@aivarannamaa you should specify it in the manifest instead (or also), see https://github.com/rm-hull/luma.core/blob/master/MANIFEST.in for an example.

aivarannamaa, 2020-01-15:
If I'm not mistaken, MANIFEST.in controls the list of files present in [tarball and wheel](https://pypi.org/project/thonny/#files). I suppose `__pycache__` is not present there

thijstriemstra, 2020-01-15:
right, the place where @stdedos found the files.

aivarannamaa, 2020-10-28:
Seems to be fixed in 3.3b2

closed 2020-10-28 by aivarannamaa