thonny/thonny #1061: Thonny Birdseye debugger: localhost\7777\ipython_call refuses to connect.

HEELcaT666, 2020-01-14:
I was trying out the Birdseye debugger on Thonny and I have stumbled upon this issue where it does not open up the server at all. I checked using `netstat -ano` that none of the ports were used up. I tried running this simple python program: 
```python
from birdseye import eye

@eye
def hello():
    lst = ["hello%d" % i for i in range(10)]
    for i in lst:
        print(i)
        
if __name__ == "__main__":
    hello()
```
on Thonny using Birdseye and it does not work. I have tried running birdseye form console using `python -m birdseye` from both my own version of python (3.8.1) as well out of Thonny's native directory, which both work. I also tried using birdseye in other editors like VS Code and it worked perfectly. I have no idea why this may be happening. Any ideas what the bug is?

aivarannamaa, 2020-01-19:
Thank you for the report!

I could not reproduce in the development environment, but I could reproduce this in installed Thonny. It looks like it has to do with Thonny's non-standard console allocation. The stacktrace in frontend.log was following:

```
ERROR: Problem running Birdseye server
Traceback (most recent call last):
  File "C:\Users\Aivar\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\birdseye_frontend.py", line 23, in start_server
    port=get_workbench().get_option("run.birdseye_port"), debug=False, use_reloader=False
  File "C:\Users\Aivar\AppData\Roaming\Python\Python37\site-packages\flask\app.py", line 938, in run
    cli.show_server_banner(self.env, self.debug, self.name, False)
  File "C:\Users\Aivar\AppData\Roaming\Python\Python37\site-packages\flask\cli.py", line 629, in show_server_banner
    click.echo(message)
  File "C:\Users\Aivar\AppData\Roaming\Python\Python37\site-packages\click\utils.py", line 260, in echo
    file.write(message)
AttributeError: 'NoneType' object has no attribute 'write'
```

aivarannamaa, 2020-01-19:
It looks like this is duplicate of #847 

Unfortunately, at the moment I don't have any other ideas besides waiting until click projects gets this thing sorted out (or starts offering more flexible api).

closed 2020-01-19 by aivarannamaa