thonny/thonny #1062: Labels for toolbar buttons

mixmastamyk, 2020-01-15:
Please consider adding text labels to the toolbar buttons, or an option at least.  They help users a lot since small pictures are often not enough to remember usage.  Another benefit is that it will give them some space, the buttons are small and closely packed to the left, making them hard to hit.

Thanks for the cool editor!

aivarannamaa, 2020-01-15:
Did you try "Simple mode" (Tools => Options => UI mode)? This mode has larger buttons and button captions. I expected that in regular mode one uses keyboard shortcuts.

mixmastamyk, 2020-01-16:
No I didn't, that's cool.  Thanks.  

However it didn't start up in simple mode, how will a beginner figure that out?  And what else is tied to this UI mode?  Usually this option is changeable on its own.

aivarannamaa, 2020-01-16:
[Simple mode](https://github.com/thonny/thonny/blob/master/thonny/plugins/help/modes.rst) was included in Thonny by the wish of Raspberry Pi people. In Raspberry this is the default mode. In regular distributions I'm using regular mode as default, because IMO this provides more flexibility without noticeable extra complexity.

I may rethink this in a future version and make the simple mode default.

Did you notice that in regular mode the toolbar buttons have tooltips?

mixmastamyk, 2020-01-16:
Yes I did.  I installed it under the latest Ubuntu.

Thanks,

closed 2020-01-16 by mixmastamyk