thonny/thonny #1064: Strange response from Thonny

Ferrangie, 2020-01-17:
This is the line I input to the REPL of Thonny:
test_code = "\N{SNAKE}"

and this is the response when I press Enter/Return:

"Problem when handling 'TopLevelResponse"
Traceback (most recent call last):
  File "c:\users\mrwig\appdata\local\programs\python\python37\lib\site-packages\thonny\workbench.py", line 1582, in event_generate
    handler(event)
  File "c:\users\mrwig\appdata\local\programs\python\python37\lib\site-packages\thonny\plugins\variables.py", line 64, in _handle_toplevel_response
    self.show_globals(event["globals"], "__main__")
  File "c:\users\mrwig\appdata\local\programs\python\python37\lib\site-packages\thonny\plugins\variables.py", line 71, in show_globals
    self.update_variables(globals_)
  File "c:\users\mrwig\appdata\local\programs\python\python37\lib\site-packages\thonny\memory.py", line 113, in update_variables
    self.tree.set(node_id, "value", description)
  File "c:\users\mrwig\appdata\local\programs\python\python37\lib\tkinter\ttk.py", line 1482, in set
    res = self.tk.call(self._w, "set", item, column, value)
_tkinter.TclError: character U+1f40d is above the range (U+0000-U+FFFF) allowed by Tcl

Ferrangie, 2020-01-17:
The preceding post also seems to set a condition in Thonny whereby (for example) a subsequent input:
"""This string has double " and single
... quotes ' inside of it"""
will result in a simliar (same?) TopLevelResponse error.

HOWEVER, running a short program like: 
import random
random.seed(100)
actual_value = 0
truncated_value = 0

def truncate(n):
    return int(n*1000)/1000

for _ in range(10000):
    randn = random.uniform(-0.05, 0.05)
    actual_value = actual_value + randn
    truncated_value = truncate(truncated_value + randn)

print("Actual: ", actual_value)
print("Trunc: ", truncated_value)

(end of code)

will clear the condition, and repeating the input of:
"""This string has double " and single
... quotes ' inside of it"""

properly results in:
'This string has double " and single\n... quotes \' inside of it'

with no TopLevelResponse error.
Inputing the original offending code: test_code = "\N{SNAKE}" results in a repeat of the error cycle.

aivarannamaa, 2020-01-18:
Which Thonny version are you using? There were some Unicode-related updates in the last release

Ferrangie, 2020-01-25:
The update to 3.2.6 (from 3.2.4) did fix this problem.

closed 2020-01-25 by Ferrangie