thonny/thonny #1065: Thonny IDE missing its Menu Bar

BazzF, 2020-01-18:
I am new to the Raspberry Pi and am learning Python.

I have a problem with the Thonny Python IDE, the menu bar at the top of the window (File Edit View Run Device Tools Help) is missing and I don't know how to get it. I don't ever recall it being there in the first place and only discovered it's absence when I was looking at some code examples on Google.
I've tried sudo apt full-upgrade to update all programs but the menu bar is still missing.
I think I am running version 3.2.6-1 (I saw this revision number whilst running the full-upgrade

How do I get the Menu Bar to appear?

aivarannamaa, 2020-01-18:
Raspberry uses simple mode as default for Thonny. You should have a text like "Switch to regular mode" in the upper-right corner of the window. This will give you menu bar and other features.

BazzF, 2020-01-18:
Thanks for your response. At first I couldn't find this text at first, but
the resolution my monitor was set to was incorrect and the text you
mentioned wasn't on screen. I changed the resolution, found the text,
clicked on it, restarted Thonny and the menus appeared.

Thanks again for your help.

closed 2020-01-18 by BazzF

mrrobot0123, 2020-08-20:
I have the same problem . What was the resolution that you reset to?

joetaylor010, 2021-02-10:
Same issue. Going full-screen (only tested 1440p) allowed me to see the option tucked in the top right corner. Users with lower resolution screens may not be able to see the option at all.

aivarannamaa, 2021-02-10:
@joetaylor010, were you able to see the toolbar buttons (New, Open, etc)? Was the "Switch to regular mode" link hidden behind the upper or right edge of the monitor? Or were the toolbar buttons hiding it?

mrrobot0123, 2021-02-10:
I've got it.
Thanks for the help :)

TechWilk, 2022-08-27:
The link wasn't visible to me either at the default window size, it was hidden off the edge of the window on the right.
I was able to uncover it by resizing the Thonny window, stretching it out wider to the right.

aivarannamaa, 2024-04-18:
@Chaosdemon16, on macOS the menu should be always available. Can you post the screenshot of your whole screen?