thonny/thonny #1070: Error when trying to read directory on ESP8266

riteman, 2020-01-24:
Traceback (most recent call last):
  File "C:\Users\ritem\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1594, in event_generate
    handler(event)
  File "C:\Users\ritem\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 875, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "C:\Users\ritem\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 318, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable

sinayg, 2020-06-04:
Same here: 
can't see the directory and when trying to upload a file:
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
AttributeError: 'module' object has no attribute 'size'
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
  File "<stdin>", line 2, in <dictcomp>
AttributeError: 'module' object has no attribute 'size'
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\sinay\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\sinay\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 729, in _cmd_get_dirs_child_data
    root_data = {name: {"kind": "file", "size": size} for (name, size) in sizes.items()}
AttributeError: 'NoneType' object has no attribute 'items'

aivarannamaa, 2020-06-04:
@sinayg, which device are you using?

sinayg, 2020-06-19:
Esp32 via USB

Sinay +972-545-506-412

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you're still having problems

closed 2021-01-19 by aivarannamaa

sinayg, 2021-01-20:
thank you. will try
Sinay
+972 545 506 412