thonny/thonny #1072: Interrupt running program

cornwarecjp, 2020-01-25:
I've been looking for an IDE to help people learn programming. My benchmark is QBasic, which helped me learn programming. Most environments are way too complex or don't offer the right feature set. Thonny comes close, and even offers some nice features I didn't think of.

One thing I've been missing though is the ability to interrupt a running program. In QBasic, this was done with the Ctrl+Break key combination, in Thonny it makes more sense to have a "pause" button in the Thonny toolbar. In an interrupted program, users should be able to do typical debugging things:

* Inspect variables
* Inspect the calling stack
* Step-wise continuing through the program
* Continue the program
* Stop the program

Final note: there are many IDEs that offer this feature, so maybe it's already there and I just overlooked it. I'm using v3.1.2 and I didn't see it, and I didn't see it added later in the changelog. If it already exists, this issue changes to making this feature easier to find.

aivarannamaa, 2020-01-25:
Thonny allows interrupting programs with Ctrl+C. It doesn't work in some cases, though (eg. when program is doing IO or sleeps). I intend to enhance this under #922.

Pause (with resume) would be more difficult to implement. What would be the use case of it compared to setting a breakpont in advance?

cornwarecjp, 2020-01-25:
Use case: a program is not doing what you expect (e.g. it hangs somewhere in a busy loop), and you want to know what's going on. Pause with resume allows you to debug it immediately, instead of a trial-and-error process of setting breakpoints, adding print statements and so on.

aivarannamaa, 2020-02-10:
I need to experiment with catching KeyboardInterrupt to see whether this is feasible.