thonny/thonny #1073: Run interactive Python commands while debugging

cornwarecjp, 2020-01-25:
I've been looking for an IDE to help people learn programming. My benchmark is QBasic, which helped me learn programming. Most environments are way too complex or don't offer the right feature set. Thonny comes close, and even offers some nice features I didn't think of.

One thing I've been missing though is the ability to run interactive Python commands while debugging. In QBasic, the "Direct" sub-window made this possible. It should be possible to access and/or change variables of the running process using interactive commands.

I can imagine this is a difficult feature to implement. For instance, when a program is interrupted in the middle of a function call, in which context is the interactive command executed? If it is possible to choose a context in which to run an interactive command, it might make sense to offer a "command" sub-window in every context window. When using multiple modules, maybe the context in the main window is the global context of whatever module tab is active in the IDE? Should be the same behavior as e.g. in the Variables sub-window.

I'm not sure if this is a necessary consequence, but in my view, interactive commands should be separate from the "Shell" sub-window, which then becomes only for providing input/output to whatever code is running. This makes it clear whether you're providing commands or input.

More integration of debugging and interactive commands gives rise to the question when the variable namespace should be reset. For a function it's clear: it's deleted whenever the function is left. For the top-level it's not so clear, especially if you can continue to use interactive commands while not running a program. I guess that, by default, the global namespace should be emptied whenever restarting a program: leaving dirt from previous runs (and interactive commands) leads to different behavior on different runs, which will be confusing for users.

aivarannamaa, 2020-01-25:
I'm commenting last topic first -- F5 always runs programs from clean environment.

I've considered adding such feature, but I'm afraid it would complicate the UI without bringing noticeable benefits. I agree that evaluating expressions could help in debugging, but the Object inspector offers pretty good support for this. I don't see why one would want to change the variables during debugging.

Don't forget, that Thonny's niche is being a simple and useful tool for learning programming.

aivarannamaa, 2021-01-19:
Discussion resurfaced under #1520

closed 2021-01-19 by aivarannamaa