thonny/thonny #1074: Debugging by default

cornwarecjp, 2020-01-25:
I've been looking for an IDE to help people learn programming. My benchmark is QBasic, which helped me learn programming. Most environments are way too complex or don't offer the right feature set. Thonny comes close, and even offers some nice features I didn't think of.

One thing I've noticed though is that Thonny offers multiple ways of starting a program, and running in a debugger doesn't "feel" like it's offered as the default. I think, for beginning programmers, it makes much more sense to always start in a debugger. It may be slower, but that's a secondary concern, after being able to see what a program is doing. Once you're confident your code is correct, you can always choose one of the non-default options to run your program a bit faster.

So I'd suggest to have a "play" button in the toolbar that starts the program in the nicest debugging mode you can offer. All other (faster) running options should only be in the (Run) menu, so the toolbar isn't cluttered with options that may be confusing to beginning users.

aivarannamaa, 2020-01-25:
In most IDE-s, invoking "Debug" without any breakpoints would run the whole program. This makes the command usable for just running.

Thonny takes different approach. If there are no breakponts, debug commands will stop before first statement. This allows introducing stepping without talking about breakponts first. Therefore "Run" isn't just a faster variant of "Debug" in Thonny.

cornwarecjp, 2020-01-25:
Without immediately jumping to conclusions on what's right and what's wrong, let me describe QBasic's way. I think it has three "states"(*): not-running, running and interrupted. Key presses cause state changes:

not-running + F5 -> running
not-running + F8 -> interrupted (at start of program)

running + Ctrl+Break -> interrupted (wherever it was at that moment)
running + "exception" -> interrupted (wherever it was at that moment)
running + end of program -> not-running

interrupted + F5 -> running (continuing where it was interrupted)
interrupted + F8 -> interrupted (after executing a single step)
interrupted + ?? -> not-running

This model allows you to step, either right from the start or after interrupting a running program, without using breakpoints. I guess it makes most sense in conjunction with #1072 though.

In the end, you still have two "buttons": when not-running, F5 and F8 in QBasic roughly correspond to "Run" and "Debug" in Thonny. Intuitively I think the "Debug" button can be merged with one of the step buttons: when not-running, pressing that button just makes you perform the very first step of the program (or the "zero'th step", which is starting the program in the first place). Making the "Run" button start the program in a debugger is mostly useful for enabling #1072; not much else AFAIK.

The most important thing to gain here is to avoid having to explain the difference between running and debugging, at least early on. In the beginning, it's just the same thing. I think the "video" metaphor is very powerful, as most people are already familiar with "play", "pause" etc. symbols. As a beginner, I don't know what that beetle button is about or what debugging is, but when a program is not-running and a button says it performs a single step, I immediately assume it performs only the first step of the program, and leaves the program paused after that. So that should be intuitive for a beginner.

(*) Of course these are really sets of states: running can mean running anywhere in the program, with any possible program state.

aivarannamaa, 2020-02-10:
This is interesting. What about just enabling step (into?) command when program is not running. Does QBasic differentiate between "step over" and "step into"?

Sorry for not replying earlier -- I was really busy with other stuff and I didn't want to deal with this issue in a hurry. I really appreciate your input -- these are very interesting questions for me as well and I have postponed them only because they require time and attention.