thonny/thonny #1075: Change program code while debugging

cornwarecjp, 2020-01-25:
I've been looking for an IDE to help people learn programming. My benchmark is QBasic, which helped me learn programming. Most environments are way too complex or don't offer the right feature set. Thonny comes close, and even offers some nice features I didn't think of.

One thing I've been missing though is the ability to change program code while it's runnig (interrupted in a debugging session). QBasic can do it, why can't Thonny? I can imagine it might be difficult to convince the Python interpreter to change running code though - I don't know the details.

I am aware this is similar to issue #400, but from what I understand in that discussion, that's only about changing variables, not about changing code. Therefore, I hereby create a new feature request, for the "changing the code" feature.

aivarannamaa, 2020-01-25:
I'm afraid the possibility to change the code during debugging would complicate the UI (and the understanding of Python's notional machine) without noticeable benefits to the learning process.

cornwarecjp, 2020-01-25:
I can see it's difficult to implement and maybe it has little added value. I don't fully understand what you mean with "the understanding of Python's notional machine".

I guess it's mostly a productivity feature: you can fix bugs as you see them happening, and see if your fix works, all without having to restart your program. I still think it would be cool, but maybe not the highest priority when it comes to building an educational tool.

aivarannamaa, 2020-02-10:
>  I don't fully understand what you mean with "the understanding of Python's notional machine".

For me the priority has been to explain the core principles of Python execution as clearly as possible. And this is easier to do if the program is fixed during the execution process.

I agree that it more dynamic approach may be more convenient, but IMO this would be better fit for Pycharm and other professional IDE-s.

closed 2021-01-19 by aivarannamaa