thonny/thonny #1076: Error when saving file to ESP8266

riteman, 2020-01-25:
[image]

the terminal window shows

PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 743, in _cmd_write_file
    self._write_file(generate_blocks(cmd["content_bytes"], BUFFER_SIZE), cmd["path"])
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1094, in _write_file
    result = self._write_file_via_serial(content_blocks, target_path, notifier)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 1166, in _write_file_via_serial
    self._execute_without_output(script)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 446, in _execute_without_output
    raise RuntimeError("Failed MP script: " + str(out) + "\n" + str(err))
RuntimeError: Failed MP script: 
Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

  File "<stdin>", line 4, in __W

OSError: 28

Version is 3.2.6

Best
Lars

aivarannamaa, 2020-02-10:
Thank you for the report and sorry for not replying earlier.

Does it happen all the time? Can you give me a link to the exact device you are using?

riteman, 2020-02-10:
Hi Aivar

 

It does not occur that often – I just thought I would let you know in case you had an idea for a fix.

However, I sometimes see the system freeze when I update files on the chip – after I restart Thonny, the chip then is empty (no files)

 

The board is a “standard” Wemos D1 mini:

Best regards

Lars

 

PS: Do you have any plans to add a very basic version control to Thonny? – that is the functionality I miss the most

riteman, 2020-02-10:
Here I tried to save a py file into the lib directory and Thonny crashes without reason:

 

Best regards

Lars

aivarannamaa, 2020-02-10:
> PS: Do you have any plans to add a very basic version control to Thonny? – that is the functionality I miss the most

Nope. I may add simple "Import from GitHub" and "Export to GitHub", but I'll leave more serious version control for other apps.

BTW: GitHub doesn't show attachments from e-mail. It looks like you tried to send some images, but I don't see them neither in my e-mail client nor in GitHub.

riteman, 2020-11-05:
Hi

I have the latest beta version of Thonny installed and added birdseye to the
package.
I added @spy to my program but I am unable to start the debugger - I have
watched Youtubes and read a lot on the web around debugging with Thonny but
everybody starts where the debugger buttons are active.

Any help will be much appreciated!

Best regards
Lars

riteman, 2020-11-05:
Forgot one important fact: 

I am running micropython on ESP32's

Lars

_____________________________________________
From: riteman2@gmail.com <riteman2@gmail.com> 
Sent: 5. november 2020 19:27
To: reply@reply.github.com
Subject: Debug using Birdseye

Hi

I have the latest beta version of Thonny installed and added birdseye to the
package.
I added @spy to my program but I am unable to start the debugger - I have
watched Youtubes and read a lot on the web around debugging with Thonny but
everybody starts where the debugger buttons are active.

Any help will be much appreciated!

Best regards
Lars

aivarannamaa, 2020-11-05:
Neither Birdseye nor any of the other Thonny's debuggers work with MicroPython.

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you're still having problems

closed 2021-01-19 by aivarannamaa