thonny/thonny #1077: debug does not work

datensysteme, 2020-01-25:
I would like to work with thonny. Editor looks good, Run works, but debug does not work. Klick on the bug symbol is like running the script, nothing is changing, no possible single step etc...

Dont know what to do

Walter

aivarannamaa, 2020-01-25:
Do you have breakpoints in any of your open files?

datensysteme, 2020-01-25:
Hi,

I tried everything concerning debug…

I want to debug only one single .py-file, which can be debugged within Pycharm, that means, nothing is wrong.

When I use debug in thonny, it starts with a yellow line at the beginning (that is ok) and steps through the code but dont step into funktions

Greetings
Walter

aivarannamaa, 2020-01-26:
Are the functions defined in your own script or in a library? Thonny does not step into library functions, but this can be changed in Tools => Options => Run & Debug.

datensysteme, 2020-01-26:
Functions are within my own script.

Structur is:
import…

import…

…

def start_script():

scan_input()

def scan_input():

def convert_to_jpg(parameter):

def delete_images():

…

def start():

start_script()

start()

This script loops like the “Server”-part of a client-server-software. I can debug it in pycharm, but in thonny I cannot enter into e.g. scan_input function.

aivarannamaa, 2020-01-26:
Can you post the whole script? (eg. by attaching the script in GitHub, so that the indentation doesn't get lost)

datensysteme, 2020-01-27:
Difficult…

You cannot run the script itself, there are a lot of configuration and installation around to use this system.

You might be used to make a complete testing environment.

aivarannamaa, 2020-02-27:
Did you use breakpoints for debugging it?

datensysteme, 2020-02-27:
Yes, I tried breakpoints, but thonny does not reakt on them, although the scrpt runs properly and does 100% what it should do

Greetings
Walter

aivarannamaa, 2020-02-28:
Sorry, I can't start solving the problem if I can't reproduce it.

closed 2020-11-07 by aivarannamaa

RionDsouza, 2022-08-30:
Hello I just installed Thonny. I have to same problem. Trying to debug a Python Script but the debug options just seems to run the script. Can you help?

aivarannamaa, 2022-08-31:
@RionDsouza, do you have your code in named or unnamed editor? I can't reproduce this problem when the code is in named (i.e. saved) editor, but I did notice debugger doesn't work properly if the editor is not saved (https://github.com/thonny/thonny/issues/2415).