thonny/thonny #1078: Request -- Addition of vertical indent lines --

copobotics, 2020-01-26:
Hi there, Thonny!

 I have been using this program for a while now and I find amazing everything you have done, although I am usually missing a feature most IDE have, which is an option to allow vertical indentation lines. They would help making the code way clearer, specially when you have lots of nested loops and conditions. 

Thank you for making this possible and I hope we can all enjoy some pretty indent lines sooner than later ;)

aivarannamaa, 2020-01-26:
Unfortunately I don't know how to do it in cross-platform way. I could use a trick with [bgstipple](https://www.tcl.tk/man/tcl8.6/TkCmd/text.htm#M45) in the Text widget, but it doesn't work in macOS.

copobotics, 2020-01-26:
As long as it is implemented as optional for those who don't want it and doesn't take much of your time I'd be elated to see it, even if it's not available for mac users :)

AlvaroHermida, 2020-01-26:
Yes, please, don't let Tim Cook's "management" hinder the development of Thonny.

abhijit1990, 2022-02-17:
Is it done yet?

aivarannamaa, 2022-02-18:
No, it is not yet done.

yuanyanhui, 2022-04-16:
This is a good feature to have especially for beginners.