thonny/thonny #1079: Venv interpreter soft link gets resolved in macOS

aivarannamaa, 2020-01-28:


kalhauge, 2020-10-06:
I second this :).  Alternatively, It would be nice with a choose virtual environment button instead of only being able to create a new one.

aivarannamaa, 2020-10-06:
Thanks for your suggestion! Still, I'd prefer to keep virtual env selection as it is now -- by selecting its interpreter (not its root directory). Unfortunately macOS file chooser always resolves symbolic links, so it doesn't allow selecting venv interpreter. 

I added a warning about this and made it possible to bypass the file selection dialog and enter the interpreter path directly. This inconvenience appears only once per externally created venv -- next time you can already choose it from the list.

I also took the opportunity to refactor the UI of the configuration page.

aivarannamaa, 2020-10-06:
Please reopen the issue, if you don't agree with my solution!

closed 2020-10-06 by aivarannamaa

kalhauge, 2020-10-07:
Works really well! Great job :)