thonny/thonny #1084: Not working

sund06, 2020-02-10:
Thonny 3.2.6 MAC OS
Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-02-11:
Please try this installer and tell me whether this helps: https://github.com/thonny/thonny/releases/download/v3.2.6/thonny-3.2.6-macos-installer.pkg

aivarannamaa, 2020-02-11:
Also, please tell me which macOS version are you using.

TOASTY-KING, 2020-02-12:
Had the same problem, I had to delete the Thonny application, and also moved the thonny library on to a flash drive and deleted the files off my computer. Finally restarted the computer, reinstalled thonny and it worked! Hope this helps it was hard to find solutions.

YPC05, 2020-02-13:
I am having the same error.
I have tried to download using above link and it is still not working.
I am on Mac OS Catalina

aivarannamaa, 2020-02-13:
@YPC05 , can you then please try https://github.com/thonny/thonny/releases/download/v3.2.6/thonny-3.2.6.pkg instead? This one uses slightly different approach.

YPC05, 2020-02-14:
Hello aivarannamaa
yes this has worked! Thank you very much!
Please can you tell me why what you provided worked over what is available on the Thonny website.

Thank you

aivarannamaa, 2020-02-14:
Previously Thonny was started by a Bash script inside the application bundle. This confused Catalina -- it assigned permissions to Bash instead of Thonny/Python. I changed the launch script to be a Python script and used a few other tricks I learned from macOS version of IDLE (Python's official IDE).

closed 2020-02-27 by aivarannamaa

gurugr, 2021-04-23:
not working with my high sierra. before it worked well but suddenly it has stopped to working .. no functioning the button (commands  of thonny) 
even alvarannamaa's copy also the same