thonny/thonny #1088: Version

fenris, 2020-02-17:
Downloaded latest thony v3.2.6 , however when I run thonny and check the version running v3.2.1 as per the attached screenshot 

[image]

aivarannamaa, 2020-02-17:
Please check whether you have another Thonny under your user Python packages. Open "Tools => Manage plug-ins" and look for "thonny" in the list. Where is it installed?

fenris, 2020-02-17:
@aivarannamaa thanks ! i just manually upgrade the Thony through that managed plug-in. 

[image]

closed 2020-02-17 by fenris

aivarannamaa, 2020-02-17:
@fenris -- I recommend you remove `~/.local/lib/python3.7/site-packages/thonny`. Try the "uninstall" button first. If this doesn't work, then remove it manuall together with related pip metadata from the site-packages dir. This way the `thonny` package from the main `site-package` becomes visible. Otherwise you'll need to upgrade the `--user` variant each time you install new Thonny version.

I don't know how Thonny package ended up there, probably because of some plug-in, which listed Thonny as dependency.