thonny/thonny #1090: Request: Collapse functions

ankith26, 2020-02-21:
I am using thonny, I really like it now.
This is just an idea, how about introducing function collapse feature where you can minimise the functions you do not want to see.

aivarannamaa, 2020-02-21:
Thanks for the idea!

Currently I don't consider this a priority feature for Thonny's target group, but I'll keep it in the TODO list.

mrssmcmaster, 2020-06-19:
On the contrary, I (as a teacher) would find this very useful for explaining to my students that once a function has been written, it is no longer necessary to know what it does. Being able to collapse it helps with that.

aivarannamaa, 2020-06-19:
@mrssmcmaster, good point! I never thought of it.

mjarboe5, 2020-07-07:
> Thanks for the idea!
> 
> Currently I don't consider this a priority feature for Thonny's target group, but I'll keep it in the TODO list.

Would be GREAT to have!!

ryanomatic, 2022-08-18:
+1 This would be fantastic!

rustyjuniper, 2022-10-24:
+1 I really would like to get the code folding feature. I just discovered, geany editor can help out here

palieux, 2022-11-29:
This would be great to have. My son uses Thonny and he loves it. However when learning nested if functions, we had to switch to a different IDE where we could collapse the underlying code. This helped him understand scope and nesting better.
Keep up the great work!

Splashaudio, 2022-12-28:
Would love to be able to do this. Will it happen?

aivarannamaa, 2022-12-28:
@palieux, have you tried explaining nested if-blocks using the "nicer" debugger (Ctrl+F5)? I believe this can explain the meaning of if-blocks even better than collapsing the bodies.

DIT-Jack, 2023-03-22:
ohh get on with it, add code folding

pak-man, 2023-03-24:
C´mon, we need this. After two years this very basic and quite essential feature is still missing.

aivarannamaa, 2023-03-24:
@DIT-Jack and @pak-man, c'mon, you want a feature, why don't you make a pull request? Or do you feel you've received less that you've paid for?

kingandaval, 2023-04-12:
> @DIT-Jack and @pak-man, c'mon, you want a feature, why don't you make a pull request? Or do you feel you've received less than you've paid for?

@aivarannamaa I am only just learning python now and using Thonny as my IDE but collapsing the code would be useful for people new to python. Sadly I cannot contribute to the Thonny project but maybe one day I can submit a pull request ;-)

The-Salty-Medic, 2023-12-08:
Happy December.  I would also love to see this feature added as a way to tidy up the code while I'm working on things.  I agree with a prior comment about folding up functions / lists / conditional blocks.

B0-B, 2024-01-05:
@aivarannamaa that's a weird argument, you are offering an open-source software as an organization and aren't able to implement a simple but very important feature and your argument is to fork it? Why should someone use your branch then anyway?

MatrixMike, 2024-04-29:
So code folding is not here, yet

coffandro, 2024-05-09:
> @aivarannamaa that's a weird argument, you are offering an open-source software as an organization and aren't able to implement a simple but very important feature and your argument is to fork it? Why should someone use your branch then anyway?

Just curious but how do you know if this feature would be "simple", it entirely depends on how just about everything else works and is implemented. there's a low chance it'd even require a full rewrite thought most likely not but at the same time it isn't really an *essential* feature now is it?

AL-M0ND, 2024-07-05:
+1 would definitely be a great feature, would save me time scrolling around my long docstrings 😂

csunwc, 2025-03-28:
+1 I would find the code folding a super helpful feature for reading codes.

pak-man, 2025-03-28:
I gave up any hope and I am gladly using vscode + rshell instead. Works wonder, and I feel happy I abandoned that clunky piece of ... of editor.

csunwc, 2025-03-28:
@pak-man I feel your frustrations. How do you use rshell to upload codes?  Do you use the instructions here...   https://github.com/dhylands/rshell

pak-man, 2025-03-31:
Yes i used that documentation to make some custom commands for VSCode. I like rshell. It´s very robust and features like sync are very very neat. You can also script in it Python (import rshell ...) which is very convenient.

Another useful bit is [pye](https://github.com/robert-hh/Micropython-Editor/tree/master): it´s a text editor that runs on the micropython board itself. Since I learned it I could not do without. It´s just damn convenient to be able to quickly change a local file without having to upload/download. Just rsync back your changes with rshell when you´re done editing locally.

upysh is also pretty handy for dealing with local files, and ugit also when the board has IP.