thonny/thonny #1091: repeating top level error

Ferrangie, 2020-02-22:
**_A somewhat random top level error keeps popping up. I have closed and reopened Thonny several times, and Thonny works okay for awhile, then this error pops up and is persistent until Thonny is shutdown and reloaded._**

Traceback (most recent call last):
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1598, in event_generate
    handler(event)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 439, in _handle_toplevel_response
    self._discard_old_content()
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1356, in _discard_old_content
    self._clear_content(proposed_cut)
  File "C:\Users\mrwig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\shell.py", line 1361, in _clear_content
    btn_pos = float(self.index(btn))
ValueError: could not convert string to float: 

**_Using Thonny 3.2.6, Winows 10 64-bit, Python 3.7.6 32-bit, and Tk 8.6.9
while working on the following:_**

import re

text_body = """
fake text to shorten listing"""

word_list = set(sorted(text_body.split()))
print(len(word_list), "in preprocessed list.")

word_list_stripped = []
for word in word_list:
    word_list_stripped.append(word.strip(',.!?"0123456789'))

word_list_lower = []
for word in word_list_stripped:
    word_list_lower.append(word.lower())

word_list =[]
for word in word_list_lower:
    if word == "":
        continue
    else:
        word = re.sub('\.', '', word) #removing strays
        word = re.sub("\'", '', word)
        word_list.append(word)

word_list_stripped = word_list_lower = None

for word in word_list:
    print(word)

print(len(word_list), "left in processed list")

aivarannamaa, 2020-02-25:
Thank you for the report!

Can you confirm, that it happens when you have produced some long output lines which get "squeezed" by Thonny? Something like this:

[image]

?

Ferrangie, 2020-02-26:
That is correct. The error occurred sometime after the long-line output you describe.

Sent from AT&T Yahoo Mail on Android

closed 2020-10-28 by aivarannamaa