thonny/thonny #1093: ModuleNotFoundError: No module named 'numpy'

absimg, 2020-02-23:
Hello,

When I import numpy as np, I receive "ModuleNotFoundError: No module named 'numpy'".

How to install numpy?

Thanks.

aivarannamaa, 2020-02-24:
Hi!

Tools => Manage packages. Enter `numpy`, press "find package from PyPI".

closed 2020-02-27 by aivarannamaa

HGX-001, 2021-04-17:
Hi!
After I follow what you said,Tools => Manage packages. Enter numpy, press "find package from PyPI".and it install success ,but when I in Tools=>open system shell and in commend line I want to run my program it still ModuleNotFoundError: No module named 'numpy'.so i don't know how to solve this problem.Thanks.

aivarannamaa, 2021-04-17:
@HGX-001, can you copy the whole command line session for me, including the information about different Python commands that Thonny adds to the beginning. Something like this: 

```
********************************************************************************
Some Python commands in the PATH of this session:
 - python    -> /home/annamaa/python_stuff/thonny/venv37/bin/python3.7
 - python3   -> /home/annamaa/python_stuff/thonny/venv37/bin/python3.7
 - python3.7 == /home/annamaa/python_stuff/thonny/venv37/bin/python3.7
 - python3.8 == /usr/bin/python3.8
 - pip       == /home/annamaa/python_stuff/thonny/venv37/bin/pip
 - pip3      == /home/annamaa/python_stuff/thonny/venv37/bin/pip3
 - pip3.7    == /home/annamaa/python_stuff/thonny/venv37/bin/pip3.7

********************************************************************************
annamaa@a2:~ $ python3 Desktop/dmo.py 
hello
annamaa@a2:~ $ 
```

HGX-001, 2021-04-18:
@aivarannamaa Thank you very mach,I have changed my python interpreter as Thonny itself interpreter,because before in this folder it have a other interpreter  and Thonny load this interpreter as a default interpreter ,after i changed it ,i have solved this problem.