thonny/thonny #1094: Custom layout and modify file name

Hal-L, 2020-02-24:
Hello, when I want to customize the layout, I follow the tutorial, and I find that I cannot find the 'data' folder. I want to add a function that can modify the file name, and its shortcut is F2. I do n’t know if this feature will be added in subsequent versions. The id can appear in order according to the execution of the code. The current appearance method is easy to see. Flower eyes. Thank you for developing this python IDE
Sorry, all of these words are in Chinese, and then Google Translate copied them for you, I hope you can understand
Finally, I wish you good health

aivarannamaa, 2020-02-25:
Thank you for the kind words! Unfortunately I could not understand your problem.

Hal-L, 2020-02-25:
Hello, this is not a problem, your compiler has greatly helped my python learning, I hope it can add a function to modify the file name, thank you.

aivarannamaa, 2020-02-25:
Have you tried "File => Rename" ?

Hal-L, 2020-02-26:
[image]
[image]
[image]Thanks, I see the file is renamed, can you add the shortcut key F2?Can the selected ids in the red box be arranged from bottom to top according to the exit order, and you can remember the column width (or you can directly set a fixed value, the fixed value of the column width is about the length of the ID), thank you, this is my little suggestion at present, and also the function I need now.Finally, thank you very much for developing this IDE. Thank you

aivarannamaa, 2020-02-27:
> can you add the shortcut key F2

I will consider this.

> remember the column width

I'll fix this.

> arranged from bottom to top according to the exit order

Did you mean *descending* order?

Hal-L, 2020-02-27:
Yes, in descending order, because when the debug presses F7, the file is too long, the ID after the end will not be seen. If it is in descending order, I press the ID after F7, it will appear in the first line on the Heap, which is more convenient to check. The order I said is the exit order of ID after F7.Thank you. Have a nice day.What's your native language, please?

aivarannamaa, 2020-10-14:
Oh, sorry, I only now noticed the question in your last comment. My native language is Estonian.