thonny/thonny #1095: Doesn't list the files if there is a thread running

bajiomeke, 2020-02-25:
When there is a thread in the code, when plug the ESP32, Thonny does not list the files from the device, even when stop/restarting it. Also it gives back this error:

Traceback (most recent call last):
  File "D:\Thonny\lib\site-packages\thonny\workbench.py", line 1598, in event_generate
    handler(event)
  File "D:\Thonny\lib\site-packages\thonny\base_file_browser.py", line 872, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "D:\Thonny\lib\site-packages\thonny\base_file_browser.py", line 315, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable

I have to try many many time Stop/Restart or even reboot my computer to get the list of the files ans then work.

bajiomeke, 2020-02-25:
If you execute this, you probably get the same problem

``
import _thread
import config
from time import sleep

def thread_Teste1():
  global var_global
  while True:
    sleep(1.5)
    config.var_global += 1
    print("Teste1: ", config.var_global)

# Thread
#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-
try:
  _thread.start_new_thread(thread_Teste1, ())
except Exception:
  import traceback
  print(traceback.format_exc())

#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-#-

while True:
    sleep(1000)
``

aivarannamaa, 2020-02-25:
Thank you for the report and example code! I'll try to reproduce this.

Which device are you using?

bajiomeke, 2020-02-25:
It's an ESP-WROOM-32, 2mb
Anything you need, hit me up.

bajiomeke, 2020-02-25:
[code-esp32.zip](https://github.com/thonny/thonny/files/4250679/code-esp32.zip)

Aivarannamaa, please, upload this code attached to the ESP32, this is what I am working on and which is causing the problem. I think it will be better cenario for your tests

DS61, 2020-03-31:
I'm having the same Issue. I successfully flashed an ESP8266-12E using a FTDI Programmer. Since then I got the same error if I try to connect to the Repl

Traceback (most recent call last):
  File "/home/jd/apps/thonny/lib/python3.7/site-packages/thonny/workbench.py", line 1598, in event_generate
handler(event)
  File "/home/jd/apps/thonny/lib/python3.7/site-packages/thonny/base_file_browser.py", line 872, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "/home/jd/apps/thonny/lib/python3.7/site-packages/thonny/base_file_browser.py", line 315, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable

If I use a Wemos D1 with Onboard USB everything is fine

aivarannamaa, 2020-09-01:
Thonny uses MicroPython REPL for communicating with the device (including querying directory contents). During this it must parse the serial output to understand when has the device finished previous command and is ready for the next. Also it must properly extract the expected response from the output. All this gets messed up when a thread keeps printing something.

I don't see any way I could support the case where management commands run by main thread and secondary threads are using same output channel.

In principle it could be possible to leave serial for only user code and set up a socket for management commands, but it would complicate matters a lot. Most likely this won't happen.

I'll update the documentation to warn against this #1315

closed 2020-09-01 by aivarannamaa

aivarannamaa, 2020-09-01:
The workaround in your case would be suspend the main thread until the secondary thread exits (eg. using a lock)