thonny/thonny #1096: Error when read a file on ESP32

ChristianFrench, 2020-02-25:
Hello,
When I "Open" a file on ESP32, there is a bug after a lopp without end :

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 969, in dispatch_from_menu
    dispatch(None)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 911, in dispatch
    handler()
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 700, in _cmd_open_file
    self.show_file(path, propose_dialog=False)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 813, in show_file
    editor = self.get_editor(filename, True)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 898, in get_editor
    return self._open_file(filename)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 885, in _open_file
    editor._load_file(filename)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 157, in _load_file
    self._load_remote_file(filename)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\code.py", line 200, in _load_remote_file
    if response.get("error"):
AttributeError: 'NoneType' object has no attribute 'get'

aivarannamaa, 2020-02-25:
Does it happen always?

ChristianFrench, 2020-02-26:
No, usually after some CTRL-C or D, some other opérations, an example without this 

[image]

GlebDen, 2020-03-21:
I've got this one error when reboot esp32:
```
Traceback (most recent call last):
  File "E:\thonny-3.2.7-windows-portable\lib\site-packages\thonny\workbench.py", line 1598, in event_generate
    handler(event)
  File "E:\thonny-3.2.7-windows-portable\lib\site-packages\thonny\base_file_browser.py", line 872, in update_dir_data
    self.cache_dirs_child_data(msg["data"])
  File "E:\thonny-3.2.7-windows-portable\lib\site-packages\thonny\base_file_browser.py", line 315, in cache_dirs_child_data
    for parent_path in data:
TypeError: 'NoneType' object is not iterable
```

aivarannamaa, 2020-10-28:
Should be fixed in the latest beta. Please reopen if still a problem

closed 2020-10-28 by aivarannamaa