thonny/thonny #1097: Error in console : need to stop and restart Thonny. Thonny is very instable with Micropython ......

ChristianFrench, 2020-02-25:
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 340, in handle_command
    response = handler(cmd)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 713, in _cmd_get_globals
    "{name : repr(value) for (name, value) in globals().items() if not name.startswith('__')}"
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 469, in _evaluate
    _, _, value_repr = self._execute_print_expr(expr, prelude, cleanup)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 466, in _execute_print_expr
    return self._execute(script, capture_output)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 441, in _execute
    return self._process_until_raw_prompt(capture_output)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 532, in _process_until_raw_prompt
    self._check_for_side_commands()
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 635, in _check_for_side_commands
    self._submit_input(cmd.data)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
Crash in backend
Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 104, in __init__
    self._mainloop()
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 140, in _mainloop
    self._submit_input(cmd.data)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')
Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 35, in read
    self._read_buffer.extend(self._read_queue.get(True, timer.time_left))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 104, in __init__
    self._mainloop()
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 140, in _mainloop
    self._submit_input(cmd.data)
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 389, in _submit_input
    echo = self._connection.read(len(bdata))
  File "C:\Users\Utilisateur\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\micropython\connection.py", line 37, in read
    raise TimeoutError("Reaction timeout. Bytes read: %s" % self._read_buffer)
TimeoutError: Reaction timeout. Bytes read: bytearray(b'')

Backend terminated or disconnected. Use 'Stop/Restart' to restart.

aivarannamaa, 2020-02-27:
Did you get this with ESP-32?

ChristianFrench, 2020-02-27:
Yes EXP32 expressif. A "[JZK® ESP-32S ESP32](https://www.jzk-jzk.com/products/esp-32s-esp32-development-board-2-4ghz-wifi-bluetooth-dual-mode-antenna-module-with-ultra-low-power)"

aivarannamaa, 2021-01-19:
Please try Thonny 3.3 and reopen if you still have problems

closed 2021-01-19 by aivarannamaa