thonny/thonny #1100: thonny themes

PsychoLeo, 2020-02-26:
Hello everyone, 

I've been working on Thonny for about 2 years (school and mostly by myself and I've started getting lazy with the white shell. I've looked up how to change the shell and it says to go in Tools --> options --> font and themes. The issue is I don't have the themes and fonts, I only have Editor/ Font/ General/ Interpreter. Could someone please help me ?

Thank you very much,
Leo

aivarannamaa, 2020-02-26:
Which Thonny version do you have?

PsychoLeo, 2020-02-27:
I'm looking at the file in which it is installed and I believe it is "jedi_0.11.1"

aivarannamaa, 2020-02-27:
Please open "Help" menu and select "About". Which version is displayed there?

Did you install Thonny with `apt` or with separately downloaded installer?

Do you see any errors when you run `thonny` from Terminal?

PsychoLeo, 2020-02-27:
I have the 2.1.16 version of thonny

PsychoLeo, 2020-02-27:
And it just says there is no such file in the terminal when I type in './thonny'

aivarannamaa, 2020-02-27:
This is really old version and themes are not implemented in it. Visit https://thonny.org for latest versions.

closed 2020-02-27 by aivarannamaa

PsychoLeo, 2020-02-27:
ok thank you very much for your help !!