thonny/thonny #1103: Harden MicroPython communication

aivarannamaa, 2020-03-02:


closed 2020-09-01 by aivarannamaa

AltoRetrato, 2020-09-01:
As always, thank you for your great work, Aivar! 😄 

I've been eagerly waiting for this, so I uninstalled v. 3.2.7 from my Windows PC and I tried [running from source](https://github.com/thonny/thonny/wiki/Running-from-the-source). Unfortunately I'm having this issue when I run it with Python 3.7.0:

`C:\Python37\python.exe: Error while finding module specification for 'thonny.__main__' (ImportError: cannot import name 'get_workbench' from 'thonny' (C:\Users\Ric\AppData\Local\Programs\Thonny\thonny\__init__.py))`

Any idea of the cause and/or solution?

Thanks!

aivarannamaa, 2020-09-01:
Did you set PYTHONPATH environment variable?

You could also try overwriting the contents of `C:\Users\Ric\AppData\Local\Programs\Thonny\thonny` with newer version and start thonny from the shortcut

AltoRetrato, 2020-09-01:
> Did you set PYTHONPATH environment variable?

Yes, I followed the instructions in [running from source](https://github.com/thonny/thonny/wiki/Running-from-the-source):

1. Downloaded https://github.com/thonny/thonny/archive/master.zip
2. Unpacked in `C:\Users\Ric\AppData\Local\Programs\Thonny`
3. Installed requirements: `pip3 install -r requirements.txt`
4. `set PYTHONPATH=C:\Users\Ric\AppData\Local\Programs\Thonny`
5. Run with `python3 -m thonny`

I also tried to `set PYTHONPATH=C:\Users\Ric\AppData\Local\Programs\Thonny\thonny`, but got the same result.

I guess I can't run Thonny from source on Windows... right?

> You could also try overwriting the contents of `C:\Users\Ric\AppData\Local\Programs\Thonny\thonny` with newer version and start thonny from the shortcut

First I downloaded `https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-windows-portable.zip` and run the `thonny.exe`, which opened the editor just fine.

I closed the editor and copied the contents of folder `thonny` from the source (master.zip) to `C:\Users\Ric\AppData\Local\Programs\Thonny\Lib\site-packages\thonny`. Now, when I run `thonny.exe` I get the same error as before, but in a GUI dialog:

[image]

AltoRetrato, 2020-09-01:
BTW, I tested the same code from https://github.com/thonny/thonny/archive/master.zip on Ubuntu, and the editor opened without any issues (but I had no time to test anything beyond that)...

aivarannamaa, 2020-09-02:
I hope to publish first beta together with binaries tonight

aivarannamaa, 2020-09-03:
@AltoRetrato , I have published Windows binaries for beta of the new version: https://github.com/thonny/thonny/releases/tag/v3.3.0b2.

It turned out the code indeed had a problem which was not apparent to me in Linux and Python 3.8