thonny/thonny #1104: Thonny does not start, The 'thonny==3.1.0' distribution was not found

wangmasterX, 2020-03-05:
Tried to uninstall and re-install thonny to get it to work with Python3.7 on my Raspberry Pi 3B+ running Raspbian, and now its does not start anymore. Nothing happens when I try to run it. Ran it through the terminal and the following error pops up:

```
Traceback (most recent call last):
  File "thonny", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3019, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3032, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 655, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 963, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 849, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'thonny==3.1.0' distribution was not found and is required by the application
```
I'm not sure why its telling me i don't have thonny when I just installed it with `sudo apt-get install python3-thonny`.

Thanks.

aivarannamaa, 2020-03-05:
Raspbian Thonny package used to be named `python3-thonny`, but now it is just `thonny`: https://www.raspberrypi.org/forums/viewtopic.php?t=248871

closed 2021-01-19 by aivarannamaa

bukhari-m, 2023-05-09:
I am using raspberry pi compute module 4s rev1.0 with debian version 10.3. I installed another python installation python3.11.3 after this however i am unable to open thonny and get the same error shown above.

[image]

aivarannamaa, 2023-05-09:
@bukhari-m, did you install Python 3.11 as system's default Python? In other words, does `python3` resolve to Python 3.11? I would advise against this, as this can break other apps as well.

It looks like your `/usr/bin/thonny` is picking up Python 3.11, but Python 3.11 does not pick up apt-installed modules (including `thonny`, and it shouldn't because apt-installed modules assume older Python version).

MobeenAshraf, 2023-06-25:
> did you install Python 3.11 as system's default Python? 

Yes-  should I downgrade?

bukhari-m, 2023-06-25:
@aivarannamaa, Yes i had made this mistake of resolving python3 to python 3.11. Now that i have corrected this. I am able to run Thonny. This time python3 resolves to the default installed python. It works now :)

MediaEU, 2024-10-08:
An ugly workaround is to change the usr/bin/thonny shebang to another python interpreter:
installed executable shebang:
#!/usr/bin/python3
changed shebang:
#!/usr/bin/python3.9