thonny/thonny #1111: Request: Connect to micropython device via webREPL instead of over serial port

olinickalls, 2020-03-11:
I have recently discovered Thonny - I can iterate my code faster than with VS code thanks to the REPL and file transfer support with devices running micropython.

The ESP8266-01 has limited pins and only a single hardware UART.  I use the UART hardware serial connection to talk to another device (an Arduino), which means disabling the UART REPL.  

    import uos
    uos.dupterm(None, 1) # kill UART REPL
    uart = machine.UART(0, baud) # init serial on UART0
    uart.init(baud, bits=8, parity=None, stop=1)

The webREPL continues as normal, though it is frustrating to have to manually reconnect after every device reset.  Further, file uploading via the webREPL is painful.

Could the webREPL with file upload be implemented in Thonny in a similar way to the normal REPL?  Understood it may have limited features compared to the serial REPL.

aivarannamaa, 2020-03-11:
Hi!
I didn't understand whether you have used Thonny's current experimental WebREPL support (since 3.2.6) and found it lacking or you are not aware of it.

olinickalls, 2020-03-11:
Wow- very cool! I didn't know about it!
Even now I can only find references in Github for api stubs from 2 years ago.  I can't see any wiki documentation or references in Thonny itself.  How can I try/configure it?
I am running 3.2.6 under Windows 10.

aivarannamaa, 2020-03-29:
Unfortunately this is not configurable. Hopefully I'll find time to update the stubs for next release

closed 2020-03-29 by aivarannamaa