thonny/thonny #1113: Menu bar, assistant, etc font super small

jussikiova, 2020-03-12:
I ran into a bit of an issue with Thonny 3.2.7 on Linux Mint 19.3. The menu bar and assistant fonts are really small and won't react to neither options/theme & font or options/general font scaling mode settings. Only setting the fonts react to is options/general UI scaling factor but that scales everything else as well. I've been using version 2.1.16 and it didn't have this problem

[image]

jiri-one, 2020-03-20:
I can confirm this issue. I have absolutely same problem on Arch Linux and on Fedora 31 too. Version od Thonny is 3.2.7
[image]

aivarannamaa, 2020-03-20:
sorry for late reply. Please check whether tweaking "Tools => Options => General => Font scaling mode" helps!

aivarannamaa, 2020-03-20:
Sorry, I now noticed that you have already tried this. I'll look into this.

jiri-one, 2020-03-20:
On Windows is it OK. Just in Linux. Thank for your answer.
[image]

Maik37, 2020-03-23:
Ich habe das selbe Problem. Ich verwende ebenfalls Linux

KTP95, 2020-04-14:
Same problem with vers 3.2.7, kubuntu 20.04 beta, and display 4k.
But... there are a fix, with UI Theme Raspberry Dark (or light) is acceptable sizing,
ps. Need restart Thonny.

But i generally with any UI Theme find another issue:

Color Patten of "MENU TOOLTIPs" is terrible, why is dark-background as all?

[image]

Maik37, 2020-04-14:
For me this didn't work. Even if I'm restarting, it's not changing.

aivarannamaa, 2020-04-14:
Have you installed Thonny with system package manager (in this case it uses system-provided Tk) or Thonny+Python binary bundle (which also contains Tk)?

Maik37, 2020-04-14:
It was already installed in my Raspberry Pi 4.

Maik37, 2020-04-14:
It was already installed on my raspberry pi 4

speedy-10, 2020-09-06:
Have the same problem as jussikovia with Ubuntu 20.04.1 (Gnome-Desktop) using 3840 x 2160 resolution.

K9Kraken, 2021-02-01:
Using Thonny on Manjaro and the font is way too small.
Thonny 3.3.3
[image]

zyuri, 2021-09-30:
Any updates on this? 
I have the same with Thonny 3.3.13 and KUbuntu 21.04...

speedy-10, 2021-10-01:
No Problems with Ubuntu 20.04, Thonny 3.3.13., UI Theme "Enhaced clam". Regards

zyuri, 2021-10-01:
When chaning the thonny script to start thonny so that it uses my system's  python 3.9 it instead of the venv works...
#!/bin/bash
# next two lines are only needed when you have an issue with BACKGROUND color
#xrdb -load /dev/null
#xrdb -query

DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )"

#export LD_LIBRARY_PATH="$DIR/../lib:$LD_LIBRARY_PATH"
export SSL_CERT_DIR=$(readlink -m "$DIR/../lib/python3.7/site-packages/certifi")
export SSL_CERT_FILE="$SSL_CERT_DIR/cacert.pem"

unset PYTHONHOME
unset PYTHONPATH
unset PYTHONSTARTUP
unset PYTHONINSPECT
#unset TCL_LIBRARY
#unset TK_LIBRARY

#"$DIR/python3.7" -E -m thonny "$@"
python3 -E -m thonny "$@"

j0fiN, 2021-10-06:
[image]

I have the same problem with Windows 10.
version: 3.3.13

How do I resolve this?

gatonero, 2022-03-30:
Same here in Manjaro
[image]

yaconsult, 2022-06-06:
Can confirm tiny, unusable menu fonts on fresh install of fedora 36 with thonny 3.3.14.  The increased and decreased font menu only affect the program windows, not the menubar.  I tried different themes as suggested above but none of them increased the size of the menu text.  The display is 1920x1200 on X11 using the latest Nvidia drivers from rpmfusion.
[image]

ErsatzMoco, 2022-06-19:
Same here, but there's a workaround: Set "Options/General/Font scaling mode" to "extra". At least in my case that restores a reasonable menu font size.

If you can't see this option item, jump to "Tools/Open Thonny data folder" and edit "configuration.ini". Find "scaling" and set it to a lower value. In the configuration file there's also "font_scaling_mode", you can set it to "extra" here as well (omit quotes, see current .ini entry).

Looks like upscaling the UI downscales the menu font in Linux or something like that, at least with font_scaling_mode = default.

yaconsult, 2022-06-23:
Correct.  I found another issue here complaining that the fonts were too large and based on the solution, I edited configuration.ini and changed "scaling = default" to "scaling = 1.0" and this fixed the problem of UI fonts being tiny.  The picture in my post above shows what thonny looked like before the edit.  This is what it looked like after:
[image]

Zaptrouiyerxi, 2023-08-09:
Nothing from above works in any combination to fix the only remaining unworkable detail for me: the size of checkboxes 

[image]

MartinRGB, 2023-09-14:
In `Ubuntu 22.04.3 LTS x86_64`

If you use `sudo snap install thonny`, you will encounter this problem, if you use `sudo apt install thonny`, you will not, and thonny versions are different.

alxxthegeek, 2023-11-18:
MartinRGB no. I installed via sudo apt install thonny and have the font problem on Ubuntu mate 23.10 with menu's only visible in dark mode and small font size for menu. That's with thonny v4.0.1

AtomicRobotMan0101, 2025-08-03:
**A Fix** 

**System:** Mint 22.1 Xia, X11, Cinnamon, Nvidia, 4608x2430 @60Hz and 275% screen scaling (I use my TV). 

Version is 4.1.7 via the `bash` script method, `apt` version and `flatpak` version were uninstalled (in frustration) and remain untested, but I don't see why this wouldn't work.

Edit `.config/Thonnyconfiguration.ini` and mod the line (about line 9) to `scaling = 10.0`

restart Thonny and fonts will be nicely sized again.

**Note:** tried other scaling factors (e.g. 6.0, 8.0, etc) and the restart of Thonny simply overrode them. Others above 10 seem OK.  I ended up on 11. I'm not bothered by the small icon size, for they are in the way anyway.

Lastly, thank you devs for Thonny. Slava Ukraini!

[image]

YorkEarwaker, 2025-12-16:
Dear @jussikiova ,

Had similar problem on Ubuntu LTS 24.04.3 Linux on a Dell laptop. Like @AtomicRobotMan0101 the general.scaling property in the configuration.ini file seemed to fix the issue.

In this instance also added it to the configuration.ini file. This item is also available via the Thonny menu. However if the menu items are too small to see without magnification then hard to change value vai the menu.

Opened configuration.ini file in Text Editor in Ubuntu added item to config and saved. Then opened Thonny and the issue was resolved. Location of the config file on Linux for the Ubuntu snap install Thonny instance is described in the link below.

Please find some notes on same at the following link. This also discusses some other property values in the config file. Thonny IDE [WS](https://github.com/YorkEarwaker/Automation/tree/main/ide/thy#issue-1) ,

[image]

Yours,
York