thonny/thonny #1115: Pillow install doesn't work with Mac OS Mojave

sharrison68, 2020-03-15:
I get the following error after trying to install pillow. I have tried on a number of Macs at our school. Those with High Sierra, it works fine, those with Mojave, the following error message exists.
[image]

aivarannamaa, 2020-03-15:
I can't reproduce this in Catalina with Thonny 3.2.7 and Pillow 7.0 or 6.2.2.

Please check whether you can install and use Pillow on independently of Thonny (eg. onto official python.org installer). If not, then you should take this to Pillow developers.

sharrison68, 2020-03-15:
Hi there,

Yes, it works fine via IDLE and pip3 install pillow. Which I did on the laptops too, but it still doesn’t work in Thonny on the Mojave systems.

Kind Regards,
Sharon

Sharon Harrison
_____________
M. Ed, B. Comp. & Inf. Science
HS Computer Science Teacher
Summer Link Computer Science Coordinator
Assistant Coach, MS Computer Science Club
Assistant Coach, HS Squash
University of Chicago Laboratory Schools
1362 E 59th Street, Chicago, Illinois 60637

aivarannamaa, 2020-03-15:
Which Thonny version and which Pillow version are you using?

sharrison68, 2020-03-15:
Thonny 3.2.7 and Pillow 7.0.0

Sharon Harrison
_____________
M. Ed, B. Comp. & Inf. Science
HS Computer Science Teacher
Summer Link Computer Science Coordinator
Assistant Coach, MS Computer Science Club
Assistant Coach, HS Squash
University of Chicago Laboratory Schools
1362 E 59th Street, Chicago, Illinois 60637

mbwolff, 2020-03-17:
I'm getting similar error messages trying to install Pygame 1.9.6 with Thonny 3.2.7 on macOS Catalina 10.15.3.

aivarannamaa, 2020-03-21:
I'm not able to reproduce either of the problems. 

You could try following options:

* uninstall pygame / pillow with "Tools => manage packages" and install them from command line (Tools => Open system shell => `pip3 install ...`). This way the packages will end up under main `site-packages` directory. 
* Try https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-xxl-3.2.7.pkg, which has both libraries preinstalled and signed by myself.

schellenberg, 2020-03-24:
Similar problems installing a library that uses Qt, using Thonny 3.2.7 on Mac OS Catalina 10.15.3. The library that I'm installing is ``cs20-easygui``, which just wraps up ``easygui_qt`` with it's prerequisites (``sip``, ``PyQt5``). When I install it via the regular pip-gui method (that I show my students), the following error appears:

[image]

If I uninstall as suggested in the comment above, and install them through the shell using ``pip3 install cs20-easygui``, I get the following error:

[image]

The relevant error above is: 
``ERROR: Could not install packages due to an EnvironmentError: [Errno 13] Permission denied: '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/sip-5.1.1.dist-info'
Consider using the `--user` option or check the permissions.``

To confirm the problem is specifically with Thonny, I used the same simple Python demo file, copied it into a new folder, then called ``pipenv install cs20-easygui``. The install worked just fine, and when the Python file was executed from within the virtual environment, everything worked as expected.

aivarannamaa, 2020-03-25:
If any of you have (or can install) separate Python 3.7 in your machine, please try `pip3 install --user ...` from regular terminal and then try importing the package from Thonny. Alternatively try installing the package from Thonny pip GUI and then use it from other Python 3.7.

Thonny's pip GUI users `--user` switch for installing, therefore with both means the package should end up in a shared location somewhere under `~/`. I'd like to know whether the problem is in Thonny's writing or reading.

schellenberg, 2020-03-25:
I installed the ``cs20-easygui`` package with the ``--user`` switch outside a virtual environment, with:

``pip3 install --user cs20-easygui
Collecting cs20-easygui
  Using cached cs20_easygui-0.9.5-py2.py3-none-any.whl (30 kB)
Collecting sip
  Using cached sip-5.1.1-cp37-cp37m-macosx_10_9_x86_64.whl (382 kB)
Collecting PyQt5
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-macosx_10_6_intel.whl (43.1 MB)
Requirement already satisfied: packaging in ./Library/Python/3.7/lib/python/site-packages (from sip->cs20-easygui) (20.3)
Requirement already satisfied: toml in ./Library/Python/3.7/lib/python/site-packages (from sip->cs20-easygui) (0.10.0)
Collecting PyQt5-sip<13,>=12.7
  Using cached PyQt5_sip-12.7.1-cp37-cp37m-macosx_10_9_x86_64.whl (63 kB)
Requirement already satisfied: pyparsing>=2.0.2 in ./Library/Python/3.7/lib/python/site-packages (from packaging->sip->cs20-easygui) (2.4.6)
Requirement already satisfied: six in ./Library/Python/3.7/lib/python/site-packages (from packaging->sip->cs20-easygui) (1.14.0)
Installing collected packages: sip, PyQt5-sip, PyQt5, cs20-easygui
Successfully installed PyQt5-5.14.1 PyQt5-sip-12.7.1 cs20-easygui-0.9.5 sip-5.1.1``

When I then attempt to run the test file in Thonny, I get the following error:
[image]

The console in Thonny has this (included as text to help search engines):
``Error importing PyQt5. Have you installed it?
Error importing PyQt5. Have you installed it?
Traceback (most recent call last):
  File "/Users/dan/Library/Python/3.7/lib/python/site-packages/easygui_qt/easygui_qt.py", line 23, in <module>
    from . import utils
  File "/Users/dan/Library/Python/3.7/lib/python/site-packages/easygui_qt/utils.py", line 126, in <module>
    class MyPageDialog(QtWidgets.QDialog):
NameError: name 'QtWidgets' is not defined

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/dan/Desktop/testing.py", line 2, in <module>
    import easygui_qt as easy
  File "/Users/dan/Library/Python/3.7/lib/python/site-packages/easygui_qt/__init__.py", line 6, in <module>
    from .easygui_qt import *
  File "/Users/dan/Library/Python/3.7/lib/python/site-packages/easygui_qt/easygui_qt.py", line 30, in <module>
    import utils
ModuleNotFoundError: No module named 'utils'``

Thonny definitely is finding the library as installed via the ``--user`` flag, since I had it completely uninstalled and got a ``module not found`` error in Thonny right before installing and re-running.

sharrison68, 2020-03-25:
> I'm not able to reproduce either of the problems.
>
> You could try following options:
>
> * uninstall pygame / pillow with "Tools => manage packages" and install them from command line (Tools => Open [...]

I uninstalled the previous Thonny, restarted my Mac, then reinstalled the version you supplied. I get the same error.

Are you trying it on Mojave OS, as that is the OS that I'm having trouble with. It works fine on High Sierra OS.

aivarannamaa, 2020-03-28:
I was able to reproduce the problem in Mojave (10.14.6):

```
>>> from PIL import Image
Traceback (most recent call last):
  File "<pyshell>", line 1, in <module>
  File "/Users/aivar/Library/Python/3.7/lib/python/site-packages/PIL/Image.py", line 69, in <module>
    from . import _imaging as core
ImportError: dlopen(/Users/aivar/Library/Python/3.7/lib/python/site-packages/PIL/_imaging.cpython-37m-darwin.so, 2): no suitable image found.  Did find:
	/Users/aivar/Library/Python/3.7/lib/python/site-packages/PIL/_imaging.cpython-37m-darwin.so: code signature in (/Users/aivar/Library/Python/3.7/lib/python/site-packages/PIL/_imaging.cpython-37m-darwin.so) not valid for use in process using Library Validation: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.
```

I also confirmed that with official Python the library is usable. 
At the moment I don't know what causes this difference, as Thonny bundles the same Python interpreter. I'll experiment with different signing options later. Maybe I need to start packaging Thonny in Mojave in order it to work both in Mojave and Catalina.

The workaround is to install separate Python from python.org and make Thonny use it as back-end (Run => Select interpreter => Alternative Python 3 ... => Known interpreters => /usr/local/bin/python3).

aivarannamaa, 2020-03-28:
Please try the installer with added entitlements: https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7_alt.exe

schellenberg, 2020-03-28:
Uninstalled the my old 3.2.7 version of Thonny (by dragging it into the trash). Then, reinstalled with the version listed above (with added entitlements). Everything worked great. The ``cs20-easygui`` package that had broken on pyQt worked flawlessly. Thanks!!

sharrison68, 2020-03-28:
That’s a windows .exe file. Do you have the Mac version?

Thanks,
Sharon

Sharon Harrison
_____________
M. Ed, B. Comp. & Inf. Science
HS Computer Science Teacher
Summer Link Computer Science Coordinator
Assistant Coach, MS Computer Science Club
Assistant Coach, HS Squash
University of Chicago Laboratory Schools
1362 E 59th Street, Chicago, Illinois 60637

aivarannamaa, 2020-03-28:
> That’s a windows .exe file. Do you have the Mac version?

I accidentally copied wrong url initially. Github comment already has the correct one: https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-ent.pkg

aivarannamaa, 2020-03-28:
> Uninstalled the my old 3.2.7 version of Thonny (by dragging it into the trash). Then, reinstalled with the version listed above (with added entitlements). Everything worked great. The cs20-easygui package that had broken on pyQt worked flawlessly.

That's great news! BTW, you don't need to remove old version before installing new -- the installer does it for you.

sharrison68, 2020-03-28:
Well, it ran, and then quit unexpectedly. I rebooted the computer, ran it again... ran okay then quit unexpectedly. It continued this a few times.

aivarannamaa, 2020-03-28:
> Well, it ran, and then quit unexpectedly. I rebooted the computer, ran it again... ran okay then quit unexpectedly. It continued this a few times.

Did the installer or Thonny itself quit? What were you doing the moment it quit?

sharrison68, 2020-03-28:
Thonny itself quit. I ran a file with this line:
from PIL import Image

It seemed to execute okay, it seemed to finish execution in the shell, and returned the next >>> but then the app closed completely. When I reopen Thonny I get the message "Thonny quit unexpectedly".

aivarannamaa, 2020-03-29:
@sharrison68 , if you can reproduce it, please check whether there is something in any of the log-files in `~/Library/Thonny`

aivarannamaa, 2021-02-27:
Please try whether https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg works better for you. It's with an experimental Universal 2 variant of Python 3.9.2.

Lochlanflipper609, 2025-01-29:
> I can't reproduce this in Catalina with Thonny 3.2.7 and Pillow 7.0 or 6.2.2.
> 
> Please check whether you can install and use Pillow on independently of Thonny (eg. onto official python.org installer). If not, then you should take this to Pillow developers.

https://pythondev.space/questions/easygui-needed-for-handling-wins-and-losses