thonny/thonny #1118: thonny crashes on shell > clear

kr-g, 2020-03-16:
thonny crashes on shell > clear

happend several times today.
no idea how to reproduce (sorry, but maybe the traceback helps...)

Traceback (most recent call last):
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/shell.py", line 1268, in _clear_shell
    self._clear_content(end_index)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/shell.py", line 1361, in _clear_content
    btn_pos = float(self.index(btn))
ValueError: could not convert string to float:

aivarannamaa, 2020-03-16:
Thank you for the report!
This seems to be duplicate of #1091

kr-g, 2020-03-16:
last time it happened during argparse. 
no "squeezed" content here.

aivarannamaa, 2020-03-16:
Alright, I leave it open then.

kr-g, 2020-03-30:
it did not occur again ...

aivarannamaa, 2020-10-28:
Duplicate of #1091

closed 2020-10-28 by aivarannamaa