thonny/thonny #1119: Windows Installer

ThomasCr, 2020-03-18:
On your [Wiki]( https://github.com/thonny/thonny/wiki/DeploymentOptions) you write, that we can use instructions from [JRSoftware](http://www.jrsoftware.org/ishelp/index.php?topic=setupcmdline) to script installation.

Problem is, that there are some problems:

/ALLUSERS should install to AllUsers - but uninstall reg key ist generated in **HKEY_CURRENT_USER**\Software\Microsoft\Windows\CurrentVersion\Uninstall\Thonny_is1

/GROUP="Informatik" should place Icon to Startmenu\Informatik\Thonny - but customization is not done (also not when use a ini file which is loaded with /LOADINF=...).

It would be nice, if this things can be fixed in upcoming versions.

Thanks,
Thomas

ThomasCr, 2020-03-18:
After some reseach on your [inno_setup.iss](https://github.com/thonny/thonny/blob/master/packaging/windows/inno_setup.iss) I found those problems:

You should place [PrivilegesRequiredOverridesAllowed](https://jrsoftware.org/ishelp/index.php?topic=admininstallmode) in your [Setup] section. See also [there](https://stackoverflow.com/questions/37221915/specify-the-registry-uninstall-key-location-hive-via-code)

This would fix all problems - but it needs a little bit more redesign your setup inno_setup.iss

aivarannamaa, 2020-03-22:
Thank you for the report and research!

I replaced some of my hacks with IS 6 features and added `PrivilegesRequiredOverridesAllowed=commandline`.

Here is the updated installer: https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7_alt.exe. Please let me know whether this works for you!

Unfortunately I could not use `PrivilegesRequiredOverridesAllowed=commandline dialog`, because it created confusing situation -- when I installed Thonny for all users first and then tried to upgrade, the installer did offer "Program files", but did not request elevation and therefore the upgrade failed. So I kept my own warnings and instructions for changing the installation mode.

ThomasCr, 2020-04-16:
Hi, today I had some time to re-validate your package:

Point 1 - the /ALLUSERS switch works now and the Uninstall-Reg-Key is written to `HKEY_LOCAL_MACHINE\SOFTWARE\WOW6432Node\Microsoft\Windows\CurrentVersion\Uninstall\Thonny_is1` where it should be placed :+1: 

But secondly changing the Folder in the Startmenu with /GROUP sadly doesn't work.

[image]

aivarannamaa, 2020-04-18:
About GROUP, I wonder whether `DisableProgramGroupPage=yes` causes this. [Documentation](https://jrsoftware.org/ishelp/index.php?topic=setup_disableprogramgrouppage) is not explicit about how it affects command line usage

aivarannamaa, 2020-04-18:
I don't have time for this at the moment, but if you have, then you can try building the installer yourself. You need to 

* install https://www.python.org/ftp/python/3.7.7/python-3.7.7.exe
* copy installed directory to `C:\Pythonny37`, such that there will be `C:\Pythonny37\python.exe` (or change build script accordingly)
* Install latest Inno Setup
* clone Thonny's repository
* open `packaging\windows`
* comment out Singtool line in inno_setup.iss (or set up your own signing)
* from Thonny repo run `create_installer.bat
* the result will be in `packaging\windows\dist`

aivarannamaa, 2021-07-24:
Thonny 3.3.12 will have refactored installer, which allows setting start-menu group both via dialog and via command line.

closed 2021-07-24 by aivarannamaa

P-Zhu, 2024-10-14:
copy ThonnyRunner310\x64\Release\thonny.exe %BUILDDIR% /Y

ThonnyRunner310\x64\Release\thonny.exe   这个程序文件 怎么生成，自己重新编译

P-Zhu, 2024-10-15:
[image]

commit db907326 (2021-07-24): Refactor Windows installer, fixes #1119 and #1909
commit 7a63cbe5 (2021-07-24): Refactor Windows installer, fixes #1119 and #1909