thonny/thonny #1120: thonny do something wrong?

GodLoveCoder, 2020-03-19:
[image]
why the al is not empty?
thonny must be do something wrong.

aivarannamaa, 2020-03-19:
Thonny does not run your programs, Python does.

I'm not sure, but I assume Python's list iterator works by indices, ie. at first loop iteration it takes the element from index 0, next time from 1 and so on until the end of the list. I recommend printing out the list in the beginning of each iteration and counting the number of iterations -- this helps to see why it is working like this.

closed 2020-03-19 by aivarannamaa

GodLoveCoder, 2020-03-19:
alright,thanks for you tell me why python did that.
but how i to finish some work like that?

aivarannamaa, 2020-03-19:
> but how i to finish some work like that?

What do you mean?

GodLoveCoder, 2020-03-19:
thanks again, I remembered I can use copy module.