thonny/thonny #1122: Feature request: PyPy Support

ankith26, 2020-03-19:
Do you plan to support pypy. AFAIK, there are no popular PyPy IDE available.

aivarannamaa, 2020-03-20:
I just experimented with latest PyPy3 on Windows and Thonny seems to be able to use it on back-end without problems. You can change back-end interpreter in "Run => Select interpreter => Alternative Python 3 ... => Locate another ... ". Note that initially the file selection dialog may display only python.exe files on Windows. You need to adjust the file filter to see pypy3.exe.

I'm looking forward to hear your feedback!

> AFAIK, there are no popular PyPy IDE available

I don't know about others, but at least PyDev (Eclipse plug-in) should work with PyPy

ankith26, 2020-03-21:
Hmm,
I saw that the file filter was at first only showing python.exe,
So it did not allow the pypy3.exe.
So I renamed the pypy3.exe to python.exe and opened it from thonnys select entepreter menu. Then thonny crashed in my system.
Then you told me about the adjust file filter to include all files.
After that, it worked perfectly.

ankith26, 2020-03-21:
Then I made a small modification
```
def _select_executable(self):
        # TODO: get dir of current interpreter
        options = {"master": self}
        if running_on_windows():
            options["filetypes"] = [
                (_("Python interpreters"), "python.exe"),
                (_("all files"), ".*"),
            ]

        filename = askopenfilename(**options)

        if filename:
            self._configuration_variable.set(filename)
```
Into
```
def _select_executable(self):
        # TODO: get dir of current interpreter
        options = {"master": self}
        if running_on_windows():
            options["filetypes"] = [
                (_("Python interpreters"), "python.exe"),
                (_("PyPy interpreters"), "pypy3.exe"),
                (_("all files"), ".*"),
            ]

        filename = askopenfilename(**options)

        if filename:
            self._configuration_variable.set(filename)
```
PS: I am messaging this from mobile so the code might not be indented properly

ankith26, 2020-03-21:
I would like to point out that thonny behaved weirdly after I made this modification. It could be due to the fact that I didn’t restart thonny after adding my modification.

closed 2020-09-16 by ankith26