thonny/thonny #1124: MYPYPATH incorrect please remove it

rwprime, 2020-03-21:
Someone please - the following error says remove it. Well, how do I do that? I did a grep -iR for mypypath but could find nothing of interest. 
On Arch VMs in Windows 10 on three PCs, I got Thonny running on two of them, but on the third one, as soon as the 'import' line is hit, I get an error in terminal. Why terminal? Well because I never got an icon in my xfce4 gui. (All 3 PCs have the same Arch build and packages)(and the given website didn't help either). Here's the error:
WARNING:thonny:MyPy: /home/richard/.local/lib/python3.8/site-packages is in the MYPYPATH. Please remove it.
See https://mypy.readthedocs.io/en/latest/running_mypy.html#how-mypy-handles-imports for more info
WARNING: MyPy: /home/richard/.local/lib/python3.8/site-packages is in the MYPYPATH. Please remove it.
See https://mypy.readthedocs.io/en/latest/running_mypy.html#how-mypy-handles-imports for more info

aivarannamaa, 2020-03-21:
What was your current directory when you got this error?

rwprime, 2020-03-22:
I'm in ~  (/home/richard)

rwprime, 2020-03-22:
Here's something I find odd. On a PC where the install of thonny had no problem, ls -lR .local/lib/python3.8/site-packages | wc - l  gives a 'word count' or line count of 211, but on the PC where the install had the error mentioned before, the line count is 7021.  What's that all about?

rwprime, 2020-03-22:
I can see where this is going. I'm probably going to just go with a cloned base build of Arch and install everything and tweak it all over again which will probably take maybe an hour and a half if I want thonny on that PC and delete the entire build with the fault.

rwprime, 2020-03-23:
So I built an entire Arch build today and did everything the same as on my other two PCs (I have a list printed out) and I got the same errors - Thonny does not show up in Start / Development with IDLE2 and IDLE3 as it does on the other two PCs and I get the following errors - oh and I put
export PATH=$PATH:/home/richard/.local/bin   in  .bashrc   and rebooted and  did another pip3 install thonny  and still I get no Thonny in Start / Development

Brand new install of Arch 3-23-20
[richard@arch-hp ~]$ pip3 install thonny
Defaulting to user installation because normal site-packages is not writeable
Collecting thonny
  Downloading thonny-3.2.7-py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 1.5 MB/s 
Collecting docutils>=0.13.1
  Downloading docutils-0.16-py2.py3-none-any.whl (548 kB)
     |████████████████████████████████| 548 kB 4.2 MB/s 
Collecting astroid>=1.4.9
  Downloading astroid-2.3.3-py3-none-any.whl (205 kB)
     |████████████████████████████████| 205 kB 5.3 MB/s 
Collecting pylint>=1.6.5
  Downloading pylint-2.4.4-py3-none-any.whl (302 kB)
     |████████████████████████████████| 302 kB 6.7 MB/s 
Collecting mypy>=0.470
  Downloading mypy-0.770-cp38-cp38-manylinux1_x86_64.whl (22.6 MB)
     |████████████████████████████████| 22.6 MB 13.2 MB/s 
Collecting jedi>=0.10
  Downloading jedi-0.16.0-py2.py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 4.8 MB/s 
Requirement already satisfied: setuptools>=33.1.1 in /usr/lib/python3.8/site-packages (from thonny) (46.1.1)
Collecting Send2Trash>=1.4.2
  Downloading Send2Trash-1.5.0-py3-none-any.whl (12 kB)
Collecting pyserial>=3.1.1
  Downloading pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB 8.5 MB/s 
Collecting asttokens>=1.1.10
  Downloading asttokens-2.0.3-py2.py3-none-any.whl (20 kB)
Requirement already satisfied: six~=1.12 in /usr/lib/python3.8/site-packages (from astroid>=1.4.9->thonny) (1.14.0)
Collecting lazy-object-proxy==1.4.*
  Downloading lazy_object_proxy-1.4.3-cp38-cp38-manylinux1_x86_64.whl (58 kB)
     |████████████████████████████████| 58 kB 3.0 MB/s 
Collecting wrapt==1.11.*
  Downloading wrapt-1.11.2.tar.gz (27 kB)
Collecting isort<5,>=4.2.5
  Downloading isort-4.3.21-py2.py3-none-any.whl (42 kB)
     |████████████████████████████████| 42 kB 846 kB/s 
Collecting mccabe<0.7,>=0.6
  Downloading mccabe-0.6.1-py2.py3-none-any.whl (8.6 kB)
Collecting mypy-extensions<0.5.0,>=0.4.3
  Downloading mypy_extensions-0.4.3-py2.py3-none-any.whl (4.5 kB)
Collecting typing-extensions>=3.7.4
  Downloading typing_extensions-3.7.4.1-py3-none-any.whl (20 kB)
Collecting typed-ast<1.5.0,>=1.4.0
  Downloading typed_ast-1.4.1-cp38-cp38-manylinux1_x86_64.whl (768 kB)
     |████████████████████████████████| 768 kB 4.5 MB/s 
Collecting parso>=0.5.2
  Downloading parso-0.6.2-py2.py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 3.6 MB/s 
Installing collected packages: docutils, lazy-object-proxy, wrapt, astroid, isort, mccabe, pylint, mypy-extensions, typing-extensions, typed-ast, mypy, parso, jedi, Send2Trash, pyserial, asttokens, thonny
    Running setup.py install for wrapt ... done
  WARNING: The script isort is installed in '/home/richard/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts epylint, pylint, pyreverse and symilar are installed in '/home/richard/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts dmypy, mypy, stubgen and stubtest are installed in '/home/richard/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed Send2Trash-1.5.0 astroid-2.3.3 asttokens-2.0.3 docutils-0.16 isort-4.3.21 jedi-0.16.0 lazy-object-proxy-1.4.3 mccabe-0.6.1 mypy-0.770 mypy-extensions-0.4.3 parso-0.6.2 pylint-2.4.4 pyserial-3.4 thonny-3.2.7 typed-ast-1.4.1 typing-extensions-3.7.4.1 wrapt-1.11.2

rwprime, 2020-03-23:
I have found the only solution that works. First I pip3 uninstall thonny and then I git clone yay an AUR helper (which I didn't want to do) and then I yay -Sy thonny  and then the Thonny icon shows up in Start / Development. This has been the only way to get that icon. I then ran a .py file and I didn't get any errors and then I tried it by starting it from terminal and ran the same file and It exited cleanly - no errors. I recommend to stop telling people to install with pip3 install thonny or pip install thonny at least if they're in Arch linux and to use yay. Using git clone doesn't work either.

aivarannamaa, 2020-03-29:
Sorry for late reply. 

When you install `thonny` with pip, then you are not supposed to get any icons -- command line (or making a desktop file yourself) is your only way. 

> "Defaulting to user installation because normal site-packages is not writeable"

This is not Thonny-specific -- in some Linuxes `pip3` tries to install into system directory and fails without `sudo`. In your case it falled back to `pip3 install --user thonny`, which is OK. I'll consider updating the recommendation and adding `--user` (but this is not required in a virtual environment).

aivarannamaa, 2021-03-31:
I could now reproduce this (it simply wasn't reproducible in my development environment) and also found the reason: https://github.com/python/mypy/issues/7604

Will include a fix in the next release.

closed 2021-03-31 by aivarannamaa