thonny/thonny #1129: "Unable to find a shell" from Tools/Open system shell

pebauer68, 2020-03-24:
Hello,
This seems to happen when x-terminal-emulator is configured as e.g. /usr/bin/terminator
bug seems to be in: def _get_linux_terminal_command():
Workaround: use other x-terminal - you need to configure with 
$sudo update-alternatives --config x-terminal-emulator
example output:
  Selection    Path                             Priority   Status
------------------------------------------------------------
  0            /usr/bin/terminator               50        auto mode
  1            /usr/bin/gnome-terminal.wrapper   40        manual mode
  2            /usr/bin/koi8rxterm               20        manual mode
  3            /usr/bin/lxterm                   30        manual mode
* 4            /usr/bin/mate-terminal.wrapper    30        manual mode
  5            /usr/bin/terminator               50        manual mode
  6            /usr/bin/uxterm                   20        manual mode
  7            /usr/bin/xterm                    20        manual mode

aivarannamaa, 2020-03-29:
Thank you for the report! I think I got it sorted out (tested also in Ubuntu 18.04). The fix will be in next release but you can also try applying the diff to your `thonny.terminal` module.

closed 2020-03-29 by aivarannamaa

aivarannamaa, 2020-03-29:
Sorry, the first commit missed one line.

pebauer68, 2020-03-29:
Thank you, I really like Thonny -  because of good documentation in the source code.

commit 45f548f0 (2020-03-29): FIXED #1129: Allow opening system shell with Terminator