thonny/thonny #1130: Installation for Python3.8.2 on Win10 with pip

Hermann12, 2020-03-27:
Why do I get this messaage after start of a new installed thonny 3.2.7 ?
Strange.

Traceback (most recent call last):
  File "c:\python38-32\lib\site-packages\thonny\workbench.py", line 544, in _start_runner
    self._runner.start()
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 102, in start
    self.restart_backend(False, True)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 557, in restart_backend
    self._proxy = backend_class(clean)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 1254, in __init__
    super().__init__(clean, get_interpreter_for_subprocess(executable))
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 1062, in __init__
    super().__init__(clean, executable)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 771, in __init__
    self._start_background_process(clean=clean)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 781, in _start_background_process
    env = get_environment_for_python_subprocess(self._executable)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 1393, in get_environment_for_python_subprocess
    overrides = get_environment_overrides_for_python_subprocess(target_executable)
  File "c:\python38-32\lib\site-packages\thonny\running.py", line 1439, in get_environment_overrides_for_python_subprocess
    if os.path.samefile(
  File "c:\python38-32\lib\genericpath.py", line 100, in samefile
    s1 = os.stat(f1)
FileNotFoundError: [WinError 3] Das System kann den angegebenen Pfad nicht finden: 'C:\\Python36-32\\python.exe'

aivarannamaa, 2020-03-27:
It looks like you have configured a Python interpreter which does not exist anymore. Please check "Run => Select interpreter"

Hermann12, 2020-03-28:
Solved for now, thanks. But where was this hidden settings? I have deinstalled Thonny and python first and installed both with updated versions.

aivarannamaa, 2020-03-28:
I'll keep this issue open for now, so I'll remember to add helpful error message when the configured interpreter doesn't exist anymore.

You'll find Thonny's configuration file when select "Tools => Open Thonny data folder".

Hermann12, 2020-03-29:
If I deinstall Thonny, there must be some old information still on the computer, otherwise why can Thonny remember the old python version? So I assume the Deinstallation is not clean enough.

aivarannamaa, 2020-03-29:
That's correct -- uninstaller does not remove Thonny's configuration directory, which is kept separate from the program directory.

I'll see whether I can enhance the uninstaller so that it allows removing the configuration as well. The difficult part is that the program can be installed system-wide but the configuration is per-user.