thonny/thonny #1132: Update MicroPython API stubs

aivarannamaa, 2020-03-29:


Josverl, 2022-10-24:
@aivarannamaa 
Would it be worthwhile to look at using/installing the [micropython-<port>-stubs](https://pypi.org/search/?q=micropython-stubs) from PyPi ?
After all, my start to build these stubs originated from Thonny code, so if anyone has a right to use them it would be you !
I generate the stubs automatically, so while they may not always have the finesse af manual crafted ones, they are up to date , and are fitted to the specific boards and versions of MicroPython.

https://github.com/Josverl/micropython-stubs#micropython-stubs
https://github.com/Josverl/micropython-stubs#micropython-stubber

aivarannamaa, 2022-10-29:
Thanks for the offer!

I have used https://github.com/hlovatt/PyBoardTypeshed/tree/main/micropython_typesheds so far. I think I'll test your stubs as well (they used to be without documentation, but not anymore, right?)

Unfortunately I can't simply use any stubs, because the `jedi` library seems to quite particular about some features used in the stub files.