thonny/thonny #1134: bug: nicer debugger overlay placement is sometimes misleading

FrankBau, 2020-03-31:
(Thonny 3.2.7 Win10 64 Python 3.7.6 Tk 8.6.9)
When I do nicer debugging (Ctrl+F5) and stepping (F7) through my simple gcd code
```
def gcd( a, b ):
    while a != b:
        if a>b:
            a = a-b
        else:
            b = b-a
    return a
g = gcd(44,12)
```
The evaluation overlay of `a != b` is shown in a wrong line:
[image]

As I want to use Thonny in my 101 prog class, thats a little bit frustrating.

aivarannamaa, 2020-04-01:
Thank you for the report! I could reproduce this and will fix it for the next release.

I'm surprised that nobody hasn't reported it earlier.

aivarannamaa, 2020-11-07:
Fix will be in Thonny 3.3

closed 2020-11-07 by aivarannamaa