thonny/thonny #1135: Code-Signing error when running pgzero with Thonny 3.2.7/MacOS Catalina

swendel, 2020-04-01:
Hi Thonny-Team,

thanks for you great tool, just started to teach my daughter Python.

We use Pygame Zero for that, therefor install "pgzero" (1.2) with Thonnys "Manage Package". 
We check the box "Pygame Zero Mode" und run a small test-program:
apple = Actor("apple")

Here's the error:
ImportError: dlopen(/Users/swendel/Library/Python/3.7/lib/python/site-packages/pygame/base.cpython-37m-darwin.so, 2): no suitable image found.  Did find:
	/Users/swendel/Library/Python/3.7/lib/python/site-packages/pygame/base.cpython-37m-darwin.so: code signature in (/Users/swendel/Library/Python/3.7/lib/python/site-packages/pygame/base.cpython-37m-darwin.so) not valid for use in process using Library Validation: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.

This problem does not appear on my other computer, where I installed Thonny 3.2.6

From "About Thonny":
Thonny: 3.2.7 (latest download)
Darwin :19.4.0 (Catalina 10.15.4)
Python: 3.7.7
Tk 8.6.8

Thanks!

Please let me know if I can help by testing something.

Greetings from Germany
Stefan Wendel

aivarannamaa, 2020-04-01:
Seems to be related to #1115.
Please tell me whether https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-ent.pkg works better for you!

swendel, 2020-04-01:
Hi Aivar,

thanks ! With your new version provided, it's working fine now, error is gone.

Greetings, Stefan.

closed 2020-04-07 by aivarannamaa