thonny/thonny #1138: Is this legal in Thonny

ray-leiter, 2020-04-05:
He is what I did:
--------------------------------------------------------------------------------
Python 3.8.2 (C:/Program Files/Python38/python.exe)
>>> a=1;print(a)
PROBLEM WITH THONNY'S BACK-END:

Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\backend.py", line 1205, in execute_source
    statements = compile(ast.Module(body=root.body[:-1]), filename, "exec")
TypeError: required field "type_ignores" missing from Module
>>> 
--------------------------------------------------------------------------------
It works in other Python shells -- is it supposed to work in Thonny?

aivarannamaa, 2020-04-05:
It does work in my set-up:

```
Python 3.7.7 (bundled)
>>> a=1;print(a)
1
>>> 
```

Which Thonny version are you using?

ray-leiter, 2020-04-06:
Aivar; 

Thanks for the reply -- I have this version: 

I  

Ray Leiter 

 

From: Aivar Annamaa 
Sent: Sunday, April 5, 2020 4:15 PM
To: thonny/thonny 
Cc: ray-leiter ; Author 
Subject: Re: [thonny/thonny] Is this legal in Thonny (#1138)

It does work in my set-up:

Python 3.7.7 (bundled)

>>> a=1;print(a)
1
Which Thonny version are you using?

aivarannamaa, 2020-04-06:
I didn't get the version. In case you intended to send an image -- this doesn't work via e-mail

ray-leiter, 2020-04-06:
Sorry – it was 3.2.4.
I guess you got my message about the update I did? (to 3.2.7)

I wish I had some idea of what to do – I don’t know what the error message is telling me about the “Backend”!

I can do any trouble shooting (run special test, etc.) we might need to figure this out.
Just let me know.
Thanks
Ray

From: Aivar Annamaa 
Sent: Monday, April 6, 2020 11:02 AM
To: thonny/thonny 
Cc: ray-leiter ; Author 
Subject: Re: [thonny/thonny] Is this legal in Thonny (#1138)

I didn't get the version. In case you intended to send an image -- this doesn't work via e-mail

aivarannamaa, 2020-04-06:
I could reproduce it with Python 3.8 set as back-end. I'll look into this, but in the meanwhile, the work-around is to use the bundled Python 3.7 instead.

ray-leiter, 2020-04-06:
Aivar;
OK, thanks for your help – I’ll use the bundled version of Python (3.7.7).
Ray

From: Aivar Annamaa 
Sent: Monday, April 06, 2020 1:42 PM
To: thonny/thonny 
Cc: ray-leiter ; Author 
Subject: Re: [thonny/thonny] Is this legal in Thonny (#1138)

I could reproduce it with Python 3.8 set as back-end. I'll look into this, but in the meanwhile, the work-around is to use the bundled Python 3.7 instead.

commit 17ddf01c (2020-04-06): FIXED #1138: Make ast.Module construction work in Python 3.8
closed 2020-04-06 by aivarannamaa