thonny/thonny #1141: birdseye

ThibaudTC, 2020-04-07:
how to get birdseye working? I am using Python 3.8.2 64 bit and the latest version of birdseye

aivarannamaa, 2020-04-11:
Thank you for the report. I'll look into this. At the moment in the development environment I get following in frontend log:

```
ERROR: Problem running Birdseye server
Traceback (most recent call last):
  File "C:\workspaces\python_stuff\thonny\thonny\plugins\birdseye_frontend.py", line 20, in start_server
    from birdseye import server
  File "C:\Py3\lib\site-packages\birdseye\server.py", line 27, in <module>
    app = Flask('birdseye')
  File "C:\Py3\lib\site-packages\flask\app.py", line 415, in __init__
    _PackageBoundObject.__init__(
  File "C:\Py3\lib\site-packages\flask\helpers.py", line 980, in __init__
    root_path = get_root_path(self.import_name)
  File "C:\Py3\lib\site-packages\flask\helpers.py", line 783, in get_root_path
    return os.path.dirname(os.path.abspath(mod.__file__))
  File "C:\Python38-32\lib\ntpath.py", line 527, in abspath
    return normpath(_getfullpathname(path))
TypeError: _getfullpathname: path should be string, bytes or os.PathLike, not NoneType

```

aivarannamaa, 2020-04-11:
In installed environment, the exception is:

```
ERROR: Problem running Birdseye server
Traceback (most recent call last):
  File "C:\Users\Maris\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\birdseye_frontend.py", line 23, in start_server
    port=get_workbench().get_option("run.birdseye_port"), debug=False, use_reloader=False
  File "C:\Users\Maris\AppData\Roaming\Python\Python37\site-packages\flask\app.py", line 985, in run
    cli.show_server_banner(self.env, self.debug, self.name, False)
  File "C:\Users\Maris\AppData\Roaming\Python\Python37\site-packages\flask\cli.py", line 670, in show_server_banner
    click.echo(message)
  File "C:\Users\Maris\AppData\Roaming\Python\Python37\site-packages\click\utils.py", line 272, in echo
    file.write(message)
AttributeError: 'NoneType' object has no attribute 'write'

```

aivarannamaa, 2020-04-11:
I assume you were using Windows. I now used different way for starting Birdseye server. You can try the solution by overwriting your birdseye_frontend.py in your installation with this: https://raw.githubusercontent.com/thonny/thonny/6e8b6e9c9ff3d9c09d2abfc46e82dc894aec79d1/thonny/plugins/birdseye_frontend.py

You can find the right directory with Tools => Open Thonny program folder => plugins. Restart Thonny after replacing the file.

ThibaudTC, 2020-04-11:
> <https://github.com/thonny/thonny/issues/1141#issuecomment-612341628>view
> it on GitHub, or

I did the overwrite as you suggested.
Still not working. Got the following:
Python 3.8.2 (P:\Python\Python38-64\python.exe)
 >>> %Birdseye array.py

Simulated Traceback (most recent call last):
   File 
"P:\Thonny\lib\site-packages\thonny\plugins\backend\birdseye_backend.py", 
line 28, in execute_source
     self._vm.load_modules_with_frontend_path(["birdseye.bird"])
   File 
"P:\Thonny\lib\site-packages\thonny\backend.py", 
line 323, in load_modules_with_frontend_path
     import_module(name)
   File 
"P:\Python\Python38-64\lib\importlib\__init__.py", line 127, in import_module
     return _bootstrap._gcd_import(name[level:], package, level)
   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
     1
   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
     1
   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
     1
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
     1
   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
     1
   File "<frozen importlib._bootstrap>", line 
219, in _call_with_frames_removed
     1
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\birdseye\bird.py", 
line 42, in <module>
     from numpy import ndarray
   File 
"P:\Thonny\lib\site-packages\thonny\backend.py", line 285, in _custom_import
     module = self._original_import(*args, **kw)
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", 
line 140, in <module>
     from . import _distributor_init
   File 
"P:\Thonny\lib\site-packages\thonny\backend.py", line 285, in _custom_import
     module = self._original_import(*args, **kw)
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\numpy\_distributor_init.py", 
line 26, in <module>
     WinDLL(os.path.abspath(filename))
   File "P:\Python\Python38-64\lib\ctypes\__init__.py", line 373, in __init__
     self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is geen geldige Win32-toepassing
     Python exception:
         OSError: [WinError 193] %1 is geen geldige Win32-toepassing

     No information is known about this exception.
     Please report this example to
     https://github.com/aroberge/friendly-traceback/issues

     Execution stopped on line 28 of file 
'P:\Thonny\lib\site-packages\thonny\plugins\backend\birdseye_backend.py'.

        26:         # Following is a trick, which 
allows importing birdseye in the backends,
        27:         # which doesn't have it 
installed (provided it is installed for frontend Python)
     -->28:         self._vm.load_modules_with_frontend_path(["birdseye.bird"])
     self: <thonny.plugins.backend.birdseye_backend.Bird...>
     Exception raised on line 373 of file 'PYTHON:\ctypes\__init__.py'.

        371:
        372:         if handle is None:
     -->373:             self._handle = _dlopen(self._name, mode)
     self: <WinDLL 'C:\Users\Dagelijks\AppData\Roaming\P...>
     global _dlopen: <built-in function LoadLibrary>
     mode: 4352
 >>>

aivarannamaa, 2020-04-12:
Does the same program work by simply running it? Or using Birdseye but with Friendly Traceback turned off?

According to the stacktrace it looks like you are trying to use 64-bit NumPy with 32-bit Python

ThibaudTC, 2020-04-12:
I tried with a program that I know that it is 
working. First just run, no Birdseye and after 
that tried to debug with Birdseye. Friendly Traceback set at 0. I got this:
 >>> %Run xy.py 52.0 5.0
  x0= 155000.0 y0= 463000.0
  dx= -26616.525556094977
  dy= -17301.196172359014
NB= 52.0 OL= 5.0
X= 128383, Y= 445698

 >>> %Birdseye xy.py 52.0 5.0
Traceback (most recent call last):
   File 
"P:\Thonny\lib\site-packages\thonny\plugins\backend\birdseye_backend.py", 
line 28, in execute_source
     self._vm.load_modules_with_frontend_path(["birdseye.bird"])
   File 
"P:\Python\Python38-64\lib\importlib\__init__.py", line 127, in import_module
     return _bootstrap._gcd_import(name[level:], package, level)
   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
   File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
   File "<frozen importlib._bootstrap_external>", line 783, in exec_module
   File "<frozen importlib._bootstrap>", line 
219, in _call_with_frames_removed
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\birdseye\bird.py", 
line 42, in <module>
     from numpy import ndarray
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", 
line 140, in <module>
     from . import _distributor_init
   File 
"C:\Users\Dagelijks\AppData\Roaming\Python\Python38\site-packages\numpy\_distributor_init.py", 
line 26, in <module>
     WinDLL(os.path.abspath(filename))
   File "P:\Python\Python38-64\lib\ctypes\__init__.py", line 373, in __init__
     self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is geen geldige Win32-toepassing
 >>>
I know that I am using Python 64bit, which is 
located at P:\Python\Python38-64\
 >>> Python 3.8.2 (P:\Python\Python38-64\python.exe)
Thibaud

aivarannamaa, 2020-04-12:
I dug deeper and found that Birdseye is trying to use `numpy` and `pandas` if they are present. 

Do I understand correctly, that the program you are trying to debug does not itself use `numpy` at all?

Do you happen to have a 64-bit NumPy installed in your user site packages (either with `pip install --user numpy` or via Thonny's manage packages)? If this is the case, then you can try whether uninstalling 64-bit NumPy helps.

Unfortunately 32-bit and 64-bit packages installed with `--user` both end up under user site packages directory of corresponding Python version.

ThibaudTC, 2020-04-13:
uninstaling numpy (64bit) and pandas did the 
trick, I didn't need them for that specific program.
Does this mean when I need numpy and/or pandas in 
a program that I cannot use Birdseye? That is not very handy.
I am using Thonny, it is simple and very handy.
Thibaud

aivarannamaa, 2020-04-13:
> Does this mean when I need numpy and/or pandas in a program that I cannot use Birdseye?

The problem was that installed NumPy was 64-bit, but Birdseye server tried to import it from 32-bit process.

There is no good solution for the 32 vs 64 bit confusion when using user site packages.

I created a pull-request for Birdseye, which would solve your case: https://github.com/alexmojaki/birdseye/pull/82

commit 6e8b6e9c (2020-04-11): FIXED #1141: Make birdseye work in Windows
closed 2021-02-27 by aivarannamaa