thonny/thonny #1144: Git integration/add-in

philwilkinson40, 2020-04-12:
This is a new feature/enhancement request.
A common skill my students need to quickly master is version control of their scripts.

- Is it possible to develop an integration/add-in with Git (or even GitHub)?

I think this was the aim of #997

aivarannamaa, 2020-04-12:
Have you tried [GitHub Desktop](https://desktop.github.com/)? IMO this is a really well designed tool which doesn't overwhelm the beginner. Beside official Windows and macOS builds, it also has unofficial and decent [Linux build](https://github.com/shiftkey/desktop/releases). I don't think trying to replicate something like this in Thonny would be a good investment of time.

On the other hand, I've been thinking about adding couple of simple Git-related commands for users who don't need to understand Git, but need to save their scripts to the cloud: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!searchin/thonny/git|sort:date/thonny/9H2FEVSQJbU/BR0NtWK0BgAJ

I'm interested in your comments about this idea.

kr-g, 2020-09-18:
i m also searching for a git plug-in and found this note here.

regarding the post on google groups and the requirement / wish of Michael Hewitt i think each student could work on a branch for each homework assignment. but that is just a first guess while reading, and most likely also not really helpful since it makes it more complicated i feel...

but is there any activity currently with respect to a git plug-in, or integration ?

aivarannamaa, 2020-09-18:
> but is there any activity currently with respect to a git plug-in, or integration ?

Nope, and I don't think I'll get to this any time soon.

kr-g, 2020-09-18:
> I don't think trying to replicate something like this in Thonny would be a good investment of time.

true, there are already other options available...

kr-g, 2021-02-24:
for a while now i m using Meld (from GNOME project -> http://meldmerge.org) for diff and such a like
- and it's quite good !
probably calling Meld (via cmd line) as external tool could be integrated somehow ?
instead of dealing with git / git-mergetool / github desktop. 
Meld sldo supports other versioning tools what is sort of good.
of course an external window is not that nice looking. 
but ... just adding this as an idea to this post ...

kr-g, 2021-02-24:
probably like context menu > open with ... 
and there an option to specify an external tool with (whatever) %F as placeholder for the file name.
similar as a linux desktop starter
just as first idea

kr-g, 2021-02-24:
and in thonny's grid-layout south panel (where shell output goes) another tab for the output of external tools...

aivarannamaa, 2021-02-26:
@kr-g, thanks for the ideas! I'll deal with external tools under #953

kr-g, 2021-11-07:
finally i wrote an own thonny plugin for git handling

plugin:
https://github.com/kr-g/thonny-gitonic

tool:
https://github.com/kr-g/gitonic

there is still a large backlog of things todo, and i dont know if it runs on all platforms (i only have linux)

but ...

Novo55, 2022-01-24:
I would like to see some access to git would be useful to me, having said that i'll have a look at gitonic maybe that will be enough