thonny/thonny #1145: Can't download using bash script on raspberrypi

samigroup, 2020-04-12:
pi@raspberrypi:~/Downloads $ bash ./thonny-3.2.7.bash

This script will download and install Thonny (thonny-3.2.7) for Linux (32 or 64-bit PC).
Press ENTER to continue or Ctrl+C to cancel.
Downloading https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-armv7l.tar.gz
--2020-04-11 22:45:01--  https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-3.2.7-armv7l.tar.gz
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-04-11 22:45:02 ERROR 404: Not Found.

pi@raspberrypi:~ $ echo "Downloading $URL"
Downloading https://github.com/thonny/thonny/releases/download/v3.2.7/thonny-xxl-3.2.7-armv7l.tar.gz
pi@raspberrypi:~ $ echo "File: :$FILENAME" 
File: :thonny-xxl-3.2.7-armv7l.tar.gz

------
Tried to manual download the file and extract, then run the install bash file, it stops at Python3. 

pi@raspberrypi:~/Downloads/thonny-xxl-3.2.7-x86_64/thonny $ ./install
Copying files to /home/pi/apps/thonny ................................ 
/home/pi/apps/thonny already exists. I need to clear it. Is it OK? [Y/n]: y
Done!
Creating start menu item (/home/pi/.local/share/applications/Thonny.desktop)  Done!
Creating Desktop shortcut ............................................ Done!
Creating uninstaller (/home/pi/apps/thonny/bin/uninstall) ............ Done!
Compiling Python files ............................................... 
[Errno 8] Exec format error: '/home/pi/apps/thonny/bin/python3.7'

Environment: 
Python 3.7.3 (/usr/bin/python3)

pi@raspberrypi:~ $ echo $(uname -m)
armv7l

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux

aivarannamaa, 2020-04-12:
Pre-built binary bundles (tar.gz) and corresponding bash script only support x86 and x64 processors.

In Raspberry you should install Thonny with `sudo apt install thonny`. 

I'll add this info to bash-script as well.

samigroup, 2020-04-12:
Other method I tried, too: 

pi@raspberrypi:~ $ sudo apt-get update && sudo apt-get install python3
Get:1 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [326 kB]
Fetched 13.4 MB in 20s (663 kB/s)                                              
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3 is already the newest version (3.7.3-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

pi@raspberrypi:~ $ sudo apt-get update && sudo apt-get install thonny
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done                              
Reading package lists... Done
Building dependency tree       
Reading state information... Done
thonny is already the newest version (3.2.6-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

Not sure what I can do to update to 3.2.7 Thonny.

samigroup, 2020-04-12:
Aivar, Thanks for your reply, 
I tried the above before reading your comment, so I tried again exactly as you mentioned it, and still gives me the 3.2.6 as latest, not the 3.2.7. 

I appreciate it, and thank you! 

pi@raspberrypi:~ $ sudo apt install thonny
Reading package lists... Done
Building dependency tree       
Reading state information... Done
thonny is already the newest version (3.2.6-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
pi@raspberrypi:~ $

aivarannamaa, 2020-04-12:
Thonny 3.2.7 only added some packaging improvements, which doesn't affect Raspberry version. In other words, on Raspberry 3.2.6 is as good as 3.2.7

closed 2020-05-24 by aivarannamaa

billhung, 2021-03-19:
Hi,
I am having this issue now. It seems like by default Thonny only goes up to 3.2.6.
The newer 3.3.3 and the altest 3.3.6 have support for Raspberry Pico, while when I tried with Thonny 3.2.6 it didn't work.
Trying to find a reliable way to upgrade on a Raspberry Pi 4 to program the Raspberry Pico

billhung, 2021-03-19:
Nevermind, I found the answer here:
https://github.com/thonny/thonny/issues/378#issuecomment-450734128
Basically did this and I got thonny 3.3.6 now. Thanks.

pip3 install thonny --upgrade
thonny -v &

PR #1202 (2020-05-24): Bash install - check architecture before downloading
Fixes #1145
commit 9d8fcd1a (2020-05-24): Bash install - check architecture before downloading (#1202)

Fixes #1145