thonny/thonny #1146: No module named pwd (Windows)

retrosnob, 2020-04-12:
Suddenly couldn't run because of this error. Reinstalled and failed, again because of this error. 

Running setup.py install for thonnyapp ... error
    ERROR: Command errored out with exit status 1:
     command: 'C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Justin\\AppData\\Local\\Temp\\pip-install-yv32s90p\\thonnyapp\\setup.py'"'"'; __file__='"'"'C:\\Users\\Justin\\AppData\\Local\\Temp\\pip-install-yv32s90p\\thonnyapp\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Justin\AppData\Local\Temp\pip-record-dd3syar8\install-record.txt' --single-version-externally-managed --compile
         cwd: C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\
    Complete output (54 lines):
    running install
    STARTING CUSTOM INSTALL
    running build
    running build_py
    creating build
    creating build\lib
    creating build\lib\thonnyappinstaller
    copying thonnyappinstaller\__init__.py -> build\lib\thonnyappinstaller
    copying thonnyappinstaller\shortcut.exe -> build\lib\thonnyappinstaller
    running install_lib
    creating C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\thonnyappinstaller
    copying build\lib\thonnyappinstaller\shortcut.exe -> C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\thonnyappinstaller
    copying build\lib\thonnyappinstaller\__init__.py -> C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\thonnyappinstaller
    byte-compiling C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\thonnyappinstaller\__init__.py to __init__.cpython-38.pyc
    running install_egg_info
    running egg_info
    writing thonnyapp.egg-info\PKG-INFO
    writing dependency_links to thonnyapp.egg-info\dependency_links.txt
    writing requirements to thonnyapp.egg-info\requires.txt
    writing top-level names to thonnyapp.egg-info\top_level.txt
    reading manifest file 'thonnyapp.egg-info\SOURCES.txt'
    writing manifest file 'thonnyapp.egg-info\SOURCES.txt'
    Copying thonnyapp.egg-info to C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\Lib\site-packages\thonnyapp-0.9.1-py3.8.egg-info
    running install_scripts
    writing list of installed files to 'C:\Users\Justin\AppData\Local\Temp\pip-record-dd3syar8\install-record.txt'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 78, in <module>
        setup(
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\site-packages\setuptools\__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\distutils\core.py", line 148, in setup
        dist.run_commands()
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\distutils\dist.py", line 985, in run_command
        cmd_obj.run()
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 69, in run
        post_install()
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 43, in post_install
        windows_post_install()
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 61, in windows_post_install
        create_windows_shortcut(os.path.join(get_windows_desktop_folder(), WIN_SHORTCUT_NAME))
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 58, in create_windows_shortcut
        '/I:"' + get_thonny_resource("thonny.ico") + '"']
      File "C:\Users\Justin\AppData\Local\Temp\pip-install-yv32s90p\thonnyapp\setup.py", line 16, in get_thonny_resource
        import thonny
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\site-packages\thonny\__init__.py", line 110, in <module>
        IPC_FILE = _get_ipc_file_path()
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\site-packages\thonny\__init__.py", line 101, in _get_ipc_file_path
        ipc_dir = os.path.join(base_dir, "thonny-%s" % getpass.getuser())
      File "C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\lib\getpass.py", line 168, in getuser
        import pwd
    ModuleNotFoundError: No module named 'pwd'
    ----------------------------------------
ERROR: Command errored out with exit status 1: 'C:\Users\Justin\AppData\Local\Programs\Python\Python38-32\python.exe' -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\Justin\\AppData\\Local\\Temp\\pip-install-yv32s90p\\thonnyapp\\setup.py'"'"'; __file__='"'"'C:\\Users\\Justin\\AppData\\Local\\Temp\\pip-install-yv32s90p\\thonnyapp\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\Justin\AppData\Local\Temp\pip-record-dd3syar8\install-record.txt' --single-version-externally-managed --compile Check the logs for full command output.

aivarannamaa, 2020-04-12:
Thanks!

I've heard about this mysterious error (#1126), but I didn't have the stacktrace so far. I'll look into this.

aivarannamaa, 2020-04-12:
It has to do with an open Python bug: https://bugs.python.org/issue32731 .
I added a workaround, which infers user name from home directory when `getpass.getuser()` fails. 

It looks like you didn't have `USERNAME` environment variable set. Any ideas why? The other guy said reboot fixed the problem for him.

retrosnob, 2020-04-12:
I'll look into it in the morning. Thanks for getting back to me so quickly.

aivarannamaa, 2020-04-15:
Do you happen to have very long (over 2047 chars) value in `PATH` environment value? (https://superuser.com/questions/237268/missing-environment-variables-in-windows-7)

commit 9b8067a5 (2020-04-12): FIXED #1146: No module named pwd (Windows)
closed 2020-09-01 by aivarannamaa