thonny/thonny #1150: Thonny doesn't like type decorations

andrastantos, 2020-04-13:
Repro:

1. Type in following code:
def my_func() -> int:
    pass

2. Press Ctrl+F5 (debug)

Expected behaviour:
    code runs without errors

Actual behaviour:
    code runs, but errors are reported:

>>> %Debug test.py
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/shared/thonny/ast_utils.py", line 178, in _mark_text_ranges_rec
    tokens = _mark_end_and_return_child_tokens(node, tokens, prelim_end_lineno, prelim_end_col_offset)
  File "/usr/lib/python3/dist-packages/thonny/shared/thonny/ast_utils.py", line 257, in _mark_end_and_return_child_tokens
    while (tokens[-1].type in (tokenize.NL, tokenize.COMMENT, token.NEWLINE, token.INDENT)
IndexError: list index out of range
Traceback (most recent call last):
  File "/home/tantos/test.py", line 1, in <module>
    def my_func() -> int:
KeyboardInterrupt: Execution interrupted
>>>

aivarannamaa, 2020-04-13:
Which Thonny version are you using?

andrastantos, 2020-04-13:
Sorry for missing that info. Version 2.1.16 on 64-bit Linux. Python 3.6.9 Tk 8.6.8.

aivarannamaa, 2020-04-14:
This is really old version. Most likely your issue was resolved in version 3.1.0. 

I assume you are using old version because your distribution (Ubutu 18.04?) doesn't have newer one. You should either apt remove it and do `bash <(wget -O - https://thonny.org/installer-for-linux)` or wait for Ubuntu 20.04, which has current version of Thonny (3.2.7)

andrastantos, 2020-04-14:
Thanks, that was it.

For people who are maybe running this in the future: this has been fixed already, only Unbuntu didn't pick up a recent enough version.

closed 2020-04-14 by andrastantos