thonny/thonny #1156: Feature request: ability to rearrange opened file tabs

hubalazs, 2020-04-14:
It would be great if we could drag and drop file tabs on the tab bar to change the order.

aivarannamaa, 2020-04-15:
Agreed.

Possibly useful resources:

* https://wiki.tcl-lang.org/page/Drag+and+Drop+Notebook+Tabs
* https://github.com/TkinterEP/ttkwidgets/blob/master/ttkwidgets/table.py

l33tlinuxh4x0r, 2020-07-06:
+1 I too would like this feature to be added to Thonny.

l33tlinuxh4x0r, 2020-12-07:
I just researched a bit into this. It looks like the following might be exactly what we need. (tkinter.dnd) https://github.com/python/cpython/blob/master/Lib/tkinter/dnd.py

l33tlinuxh4x0r, 2022-06-24:
```def make_draggable(widget):
    widget.bind("<Button-1>", on_drag_start)
    widget.bind("<B1-Motion>", on_drag_motion)

def on_drag_start(event):
    widget = event.widget
    widget._drag_start_x = event.x

def on_drag_motion(event):
    widget = event.widget
    x = widget.winfo_x() - widget._drag_start_x + event.x
    widget.place(x=x)```
This code might be useful.

mecworks, 2023-01-22:
Any status on this? Draggable tabs is a very useful feature.  I end up having to close tabs and re-open them in the order I need them which a HUGE pain.

aivarannamaa, 2023-01-23:
> Any status on this?

Still in the TODO list

closed 2023-11-26 by aivarannamaa