thonny/thonny #1157: Feature request: Copy file path to clipboard via tab context menu

hubalazs, 2020-04-14:
It would be great if the context menu on the file tabs would have another item: copy full file path to clipboard.

aivarannamaa, 2020-04-15:
I'm afraid most users wouldn't need it and it would be just extra noise for them. What would be your use case for this feature?

hubalazs, 2020-04-15:
Well, maybe you are right, beginner users probably don't need this.

use case:
several different source files are opened from different folders.
I know there is a file I would like to open that is in the same folder as one of my opened file.
I copy the filename from the tab and bring up the open file dialog, paste the file path from clipboard and then rewrite the filename part
Well in this case it would be enough to copy just the folder path, but the full filename is more general.

I know the View->Files would somewhat solve this issue

aivarannamaa, 2020-04-15:
In this case I would propose a more general (and supposedly more useful) item to the tab context menu -- "Locate in Files", which would open Files view (if not open yet), open the directory and focus the file.

aivarannamaa, 2021-01-19:
Directories can now be opened in system file browser.

closed 2021-01-19 by aivarannamaa